motioneye-project/motioneyeos #225: pi cam not detected

vidae, 2016-03-22:
hello,

My raspicam is not detected when i turn on tle system On the sd card i tried all your released (the 3 lasted),Yesterday  it's work well, but now i have a message the device is not detected.I don't understand. When i switch on the raspberry, the pi cam led light on less than 1 second. Thanks.

Gordon-1, 2016-03-26:
Could be a bad camera or a damaged RasPi (in other words a hardware problem). Did it get a static shock?  Have you tried a different RasPi Cam?  Have you tested the camera using another OS such as Raspbian?

vidae, 2016-03-26:
I actually think that there has been a waiste shock. Everything worked fine. Namely my raspberry with PICAM Modupe and a USB camera Logitech. I installed the card and raspberry camera module in a Multicomb brand case. Lacking the micro screws to secure the module after the upper shell of the case, I left it like that, the web blocking the camera in his hole. I raised the app eyemotion and everything worked for a few minutes. After I have a first error message for raspicam, then a second for the USB camera. From there the raspicam has never functioned. I tested with Raspbian and motion, with raspistill. I checked the CSI connectors, the camera module is detected, the red LED lights on but no picture is taken. I tried Raspbian relocations, motioneye etc, I still have the same result. I think maybe the components of the raspicam hit the HDMI Port causing electric shock. I do not know. I have no other raspberry to perform a test cross and hesitate to buy another raspicam ........

bortek, 2017-03-01:
If this problem is still persistent please post your question on the forum page

https://groups.google.com/forum/#!forum/motioneye

closed 2017-03-01 by bortek