motioneye-project/motioneyeos #227: RPi 3 + motionEye = slow preview/recording fps

perrychan, 2016-03-22:
First of all, great job with motionEyeOS. Was excited to see this software available and bought a couple of RPi 3s.

Unfortunately, recording stutters (1-2fps) once motion is detected. This applies whether its on large or small resolution. While watching the stream, I get 10fps, but drops down to 2-3fps as soon as it detects motion.

A lot of you don't seem to be having this problem. I am using an RPi camera when I first purchased my Raspberry Pi Model B. Think it might have anything to do with that?

ccrisan, 2016-03-22:
What fps setting have you configured? If  you're using a relatively high fps, the video encoding process might be too much for the CPU.

perrychan, 2016-03-22:
It was set to ~20fps. I have adjusted to 10fps and res to 640x480. Am getting an average framerate of 2.5 now, which is an improvement from ~1.5. It's an improvement, but it's not usable at the moment. Any ideas?

I tried to use Fast Network Camera, but the stream never appears (constant spinner).

ccrisan, 2016-03-23:
Please attach all of your log files. I'll take a look over them and see if I can spot anything fishy.

perrychan, 2016-03-24:
Attached are my logs.

[motionEyeOS-logs.zip](https://github.com/ccrisan/motioneyeos/files/187343/motionEyeOS-logs.zip)

cuddylier, 2016-03-24:
I just got a RPI 3 today as well and when it starts recording, my FPS also plummets
 even at 640 x 480 as well.

I've attached my logs.
[logs.zip](https://github.com/ccrisan/motioneyeos/files/188532/logs.zip)

cuddylier, 2016-03-24:
It appears to be any sort of recording, even with still images it starts to lag horribly and the FPS drops to 1 or 2 as @perrychan experienced.

cuddylier, 2016-03-25:
I watched the CPU usage during recording/image taking and it remains at a normal level of 30 - 40% and the I/O remains at 0% so it doesn't seem like the CPU isn't powerful enough at least?

linuscool2, 2016-03-27:
Have the same problem as @perrychan! 
Running RPi2

https://github.com/ccrisan/motioneyeos/issues/235

ccrisan, 2016-03-28:
I'll take a look over this. I haven't yet managed to investigate the problem.

linuscool2, 2016-03-28:
I will try the Motion software on Rasbian tonight to see if it's the MotionEyeOS or Motion that causes this problem. Will be back with the results

CrazyEyesPete, 2016-04-01:
Having same issue with RPI3 as well. I'm new to MotionEyeOS so maybe I'm something wrong.  When a person walks into the view, it is initially smooth video, but then looses several frames.

cuddylier, 2016-04-01:
@CrazyEyesPete Exactly the same thing I get, it's perfectly smooth for a few seconds when something moves and then drops frames as well. It still does this even with motion disabled and continuous recording enabled, it drops the frames permanently in this mode of course.

zettam, 2016-04-03:
Anyone able to find a temp solution to this, till this is fixed permanently?
I like motioneyeos but this makes it unusable :(

perrychan, 2016-04-06:
I'm hoping @ccrisan will be able to look over the logs soon.

ccrisan, 2016-04-06:
I will be taking care of this issue in the following weeks. I'm currently terribly busy with some personal matters but I haven't abandoned this project :)

ShacharWeis, 2016-04-09:
Same problem here, only much works. FPS drops to 0.1 once motion is detected. System is unusable.

ccrisan, 2016-04-09:
@perrychan , @ShacharWeis , @CrazyEyesPete , @zettam can I have your config files please? I cannot seem to reproduce the problem at all.

zettam, 2016-04-09:
I deleted everything, was waiting for the fix. Sorry.

ShacharWeis, 2016-04-09:
I've deleted the system and am now installing Zoneminder. After 12 hours my dropbox folder was full of false triggers and blank images that say "unable to open video device". Motioneyeos has a lot of potential, but I don't have time to debug it.

tsrgit, 2016-04-09:
@ccrisan , which config files and/or logs do you need?  I am having the same issue with my RPi 3.  It can easily maintain frame rates over 10 fps, even while detecting motion, as long as it is only streaming.  As soon as I enable the saving of movies onto a Samba share on a network drive, the frame rate drops to around 1 fps when motion is detected.

cuddylier, 2016-04-09:
I'm happy to provide my config as well if you let me know where to find it.

zettam, 2016-04-10:
@ShacharWeis I would not bother with zoneminder. I've tried it. It does the same thing, motion does (of course depends on what you're looking for) with a much complicated and needy setup process. It's old.

ShacharWeis, 2016-04-10:
@zettam yeah, I found that out. Can't even get Zoneminder to run at all. I've given up and am now using an old android phone. Very sad. Do you have a good guide for setting up Motion?

zettam, 2016-04-10:
@ShacharWeis motion is very easy to set up actually, did you check the online docs?
I got zoneminder running, it's not worth the effort really. Motion is much better, with the same capabilities (and a newer/better interface) with much cleaner setup.

perrychan, 2016-04-14:
@cuddylier you can find the log files under the Expert Settings menu.

cuddylier, 2016-04-14:
@perrychan I already attached my log files earlier in this issue report, just found the config backup part under General.

@ccrisan my config files: 
[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/219890/ConfigFiles.zip)

Neckbeard2016, 2016-04-15:
Same problem, currently on a RPi 2 (Model B).

linuscool2, 2016-04-15:
Is this bug only on MotionEyeOS or can switch to Motion untill the bug is fixed?

ccrisan, 2016-04-15:
I haven't been able to establish what "the bug" is about. People complain about slow fps in general. I'll take a look over the log files but I doubt there's one single problem that covers everything users are complaining about here.

linuscool2, 2016-04-17:
@ccrisan I dont know if this has something to do with the bug but read [this thread.](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=99096)

They talk about ARMv6 and ARMv7. I dont know, maybe this has something to do with it.

linuscool2, 2016-04-20:
Just installed the newest version (20160410). Seems to be working now... At least better than before.  When motion is detected the FPS drops down to ~20-22FPS.

kingsquirrel152, 2016-04-24:
I too am having the same issues. I don't see any spike in CPU load.

Kayak83, 2016-05-11:
How are you guys monitoring CPU useage? I'm running the OS image and I don't see a way of seeing the CPU temp or load. Do I need to run it in Raspbian for that functionality?

dmarkscouk, 2016-05-12:
I'm also having the same issue. I've uploaded some same video's here:
https://www.dropbox.com/s/k05uyybz5yq2nlk/stuttery.zip?dl=0

I get the same issue weather I'm running hi or low resolution, and 30 or 20 fps.
I get great video with Fast Network Camera enabled, but as I can't record this it does not help.

I'm running a Raspberry Pi 3, with the camera module 2.

Thanks, David.

ccrisan, 2016-05-13:
@dmarkscouk unfortunately I'm afraid that's the best you can get with motionEyeOS. No hardware/accelerated encoding/decoding is performed by the motion daemon and therefore the performance is limited to what the CPU can offer.

A slow SD card may impact the performance as well, while saving videos to a network share could make matters even worse.

The reason why FNA is that fast is that it doesn't use the motion "backend" but rather makes use of the streamEye project to simply capture and stream the mjpeg frames without processing them in any way (hence the great performance).

I know this is not ideal and is probably not what most of you were hoping for when decided to try motionEye(OS) but this is what we have. 

I use 7 FNA-enabled cameras based on RPI1 and a powerful central computer with 8 cores running Arch Linux + motionEye that acts as a DVR for the RPI1 cameras. I believe this is the best setup you can have with motionEye.

Phesten, 2016-05-14:
So would it not be a good idea to use a RPi 3 as a hub for 3x Foscam IP cameras? They are capable of 720p video.

I intend to offload the camera data from the Pi to a USB attached hard drive.

I really want to be able to use MotionEyeOS for this, but it won't recognize the Foscams for some reason.

tsrgit, 2016-05-17:
@ccrisan are you saying that MotionEyeOS cannot achieve better than 1 fps on a RPi when recording video with the motion engine?

ccrisan, 2016-05-18:
It can, but at lower resolutions. I remember recording 3-4 fps videos at 640x480 with RPi1.

cuddylier, 2016-05-18:
That sucks then, not really usable :(

ccrisan, 2016-05-18:
@cuddylier I totally agree. That's why I recommend using RPi1 devices just as fast network cameras. I myself have 8 of them (I just finished installing the 8th one today), all of them working in FNA mode. I use an Odroid XU4 running motionEye as a hub that performs all the motion detection, movie encoding, notifications and whatnot.

kingsquirrel152, 2016-05-18:
So - I've had some success setting the motion trigger to a low setting (it
almost seems like if the motion trigger stops - it stops recoding the
video). Ill export settings and examples later tonight.

dmarkscouk, 2016-05-19:
Would it be possible to activate video recording using a physical motion sensor or ultrasonic distance sensor (I presume this would save CPU power, that could then be used to record video)? Recording video in HD is no problem when I use the standard raspivid function.

MMiknis, 2016-05-25:
It seems to be odd, using Pi2 I can stream at 20fps once motion is detected and recording starts it drops to 0.8fps. When I check the video the first two seconds are at high frame-rate but after that fps drops :( so recorded video is no good.

hacker112, 2016-05-28:
I think motion performance could be improved by:

* Using H.264 video encoding hardware acceleration (that I think the raspberry pi has, but it is not used by motion now)

and/or

* Using threads/parallel programming, so that motion can use more that one core. The raspberry pi 3 has four cores, and one is used by motion today. I am thinking have a separate thread for video encoding and one for motion detection.

What do you think about that, @ccrisan?

ccrisan, 2016-05-28:
I totally agree that motion could make use of the GPU-accelerated routines to encode h264 movies. I believe I saw some work in this direction, although I believe it was something related to ffmpeg and not motion itself. However, I am not willing to touch motion's source code so we'll need to use the motion version that we currently have.

There are ways to use an external ffmpeg command using the *extpipe* motion option but things start to get complicated and I don't have the time to add that to the current UI and then deal with all the problems that will occur.

ikarisan, 2016-06-17:
Very disappointing. :(

I had the choice between one of these Chinese "phone home" IP cameras and the Pi3+Motion solution and I did chose the last option. :/

Now I am stuck at 352x288 videos with 10fps instead of 1280x720/30fps as offered by these chinese IP cameras. :(

ShadowNetworks, 2016-10-05:
I'm okay with setting up MotionEye on a Linux machine and catching the FNA stream. Is there any way to secure the stream? Maybe my searches didn't turn up anything.

Maxmudjon, 2016-12-01:
I have a question . I'm on raspberry PI 3 connect IP camera and watch the video broadcast but slow. You do not know the problem?

sjabbie88, 2016-12-26:
> @ccrisan That's why I recommend using RPi1 devices just as fast network cameras. I myself have 8 of them (I just finished installing the 8th one today), all of them working in FNA mode. I use an Odroid XU4 running motionEye as a hub that performs all the motion detection, movie encoding, notifications and whatnot.

> Casin, can you clarify how you have set-up motionEye on your Odroid XU4 to perform the motion detection and encoding of your 8 cameras while they are set-up as Fast Network Cameras on the Odroid XU4? 

If I imitate your setup, I only have a fraction of the FNA camera options on the MotionEye Hub device, only general settings and no motion detection or encoding options....? 

The way I read it, It is a very interesting set-up =)

ccrisan, 2016-12-26:
FNC is not available on Odroid XU4. You'll have to enable it on your RPi-based cameras, and then add them as regular network cameras to the Odroid XU4 running regular motionEye or motionEyeOS.

cuddylier, 2016-12-26:
@ccrisan You haven't seen any FPS issues using motioneyeos on your Odroid in this setup during recording of video?

ccrisan, 2016-12-26:
I run my 9 cameras at 1024x768@10fps. Provided not all of them record motion at once, things are pretty smooth.

cuddylier, 2016-12-26:
Ah, still not great then :( Not worth investing in such a system for achieving that for me sadly.

smanschi, 2016-12-30:
Since v3.1, ffmpeg offers built-in support for using the decoding/encoding capabilities of the Raspberry Pi (see for example here [http://www.cnx-software.com/2016/06/29/ffmpeg-3-1-adds-support-for-openmax-encoding-on-raspberry-pi-va-api-h-264-h-265-encoding-and-more/](url)). I am not familiar with motion or ffmpeg.. would it be possible to make use of that? E.g., ship motioneyeos with a newer version of ffmpeg and activate the hardware encoding via an option?

Here is a link with the options (At least I think they are the proper ones): [http://stackoverflow.com/questions/40175644/ffmpeg-hardware-acceleration-on-raspberry-pi](url)

KyleKatarn31, 2016-12-30:
Hello,

I'm running 3 IP Cam on RPi 1B.
Of course it's lagging a lot but i'd like to know what is the 0.3/0.0 fps that is displayed on the motioneyeOS web monitoring page ?

ccrisan, 2016-12-30:
Just hover it with the mouse and you'll see a tooltip with details. It's basically streaming/capture fps.

BigPapoo, 2017-01-08:
Disable "Motion optimization" in "Video streaming" tab.

shinji2009, 2017-02-25:
How are you guys monitoring CPU useage?

cuddylier, 2017-02-25:
Using 'top' in ssh.

shinji2009, 2017-02-25:
cuddylier wow thanks... so it is not when expert settings - Enable System Monitoring. there is some values displayed. first is temperature, last is some kind of a network speed, i don't know which and where though. but second is complete mistery, i was thinking it is cpu load, but wondering what units it is.

[24 more replies]

closed 2017-03-07 by bortek