motioneye-project/motioneyeos #228: 8192eu / D-Link DWA-131, E1 Driver

66point99, 2016-03-22:
(motionEyeOS 20160306)

I have a "D-Link DWA-131; H/W Ver.: E1; F/W Ver.: 5.00" wifi dongle - note the H/W version (not A1/B1 = 8192cu).

Can you please add "{USB_DEVICE(0x2001, 0x3319)},/* D-Link - DWA 131, E1 */ \" to the kernel?

Reference thread: https://www.raspberrypi.org/forums/viewtopic.php?p=690350

(And I'm assuming I can't simply install the 8192eu.ko referenced in that thread, as it will be specific to RPi 3.18.7-v7+, is that correct?)

Thanks & kind regards.

66point99, 2016-03-22:
Upon further digging I found: https://github.com/ccrisan/motioneye/issues/125 which probably answers my question (i.e. "No, it won't be added to MotionEyeOS - get it added to Raspbian Kernel via their processes").

Can anyone point me to a link that lists which drivers each Raspbain versions includes?

66point99, 2016-03-22:
Upon further digging I found: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=127214#p851784 which I'm hoping to install (ccrisan: I hope you don't mind me adding to this thread - I'm just trying to help myself, and others that come after me).

ccrisan, can you please state the precise RPi Kernel version that MotionEyeOs uses (Wiki says "4.1", but thread above uses numbers like "4.1.10-v7-821").

Thank you for any help you can offer.

ccrisan, 2016-03-23:
@66point99 it's 4.1.19 to be more precise. If it works with Raspbian, it *should* work with motionEyeOS as well. If for some reason it doesn't, I'll make sure it will.

66point99, 2016-03-23:
I really appreciate you input, ccrisan!

Here's what I've done so far by hybridizing from the links above (in case it's of use to someone):

SSH onto motionEyeOS RPi:
`[root@motioneye ~]# uname -a`
`Linux motioneye 4.1.18-meye #1 SMP PREEMPT Sun Mar 6 21:50:33 EET 2016 armv7l GNU/Linux`
This told me that I might be a release behind (4.1.18 rather than 4.1.19, despite using motionEyeOS 20160306 - does that sound possible?). Anyway from here on I used 4.1.18...

On my primary machine:
`wget https://dl.dropboxusercontent.com/u/80256631/8192eu-4.1.18-v7-846.tar.gz`
`scp 8192eu-4.1.18-v7-846.tar.gz root@motioneye:/home/ftp/sdcard/`

SSH onto motionEyeOS RPi:
`[root@motioneye ~]# mount -o remount,rw /`
`[root@motioneye ~]# cd sdcard`
`[root@motioneye sdcard]# tar -xzf 8192eu-4.1.18-v7-846.tar.gz`
`[root@motioneye sdcard]# install -p -m 644 8192eu.ko /lib/modules/4.1.18-meye/kernel/drivers/net/wireless/rtl8192cu`

So far, so good - but then...
`[root@motioneye sdcard]# depmod 4.1.18-meye`
`-sh: depmod: command not found`

Doh! A simple reboot of motionEyeOS RPi didn't auto-magically find my wifi dongle either ;-)

OK, I'm stuck now (as you can probably tell I don't really know what I'm doing here). Does what I've done so far seem correct? Can you advise what I need to do next?

Thanks & kind regards.

66point99, 2016-03-23:
Additional to above:
Upon looking inside the 'install.sh' script that came with the tarball, I noticed an additional command that is not mentioned in the manual instructions:
`[root@motioneye sdcard]# cp 8192eu.conf /etc/modprobe.d/.`

Doing this, and then rebooting has made no difference, however.

ccrisan, 2016-03-24:
You can't install an out-of-tree kernel driver with motionEyeOS. If it works (out-of-the-box) with Raspbian, it works with motionEyeOS as well. Otherwise, it won't.

closed 2016-03-24 by ccrisan