motioneye-project/motioneyeos #229: How do I connect to Google drive?

user150aa346, 2016-03-23:
Just to clarify which is the correct syntax from the picture below?

/My Drive/Ipsecuritycam/Video
/Ipsecuritycam/Video
https://drive.google.com/drive/folders/xxxxxxxxxxxxxxxxx
[image]

I've tried all the variations and still no luck. Thanks.

ccrisan, 2016-03-23:
/IPsecuritycam/Video?

user150aa346, 2016-03-23:
That's the syntax I'm using
[image]

ccrisan, 2016-03-23:
Isn't that a capital "P" in IPSecuritycam?

user150aa346, 2016-03-23:
It is. I've corrected it but still no success.

Regards,
Mike
------------------------------------
Michael A. McKenna
(c) 339-237-2004
I ride to support cancer research:
*You can click this link to donate to my PMC ride
<http://www2.pmc.org/profile/MM0613>*

ccrisan, 2016-03-23:
Please enable debugging in expert settings, then rerun the procedure and attach the motioneye.log file.

user150aa346, 2016-03-23:
I'll send it to you tonight. Thanks.

Regards,
Mike

>

edllew, 2016-03-25:
What works for me:
IPsecuritycam/Video
No forward slash in front or at the end.

Calduino, 2016-04-09:
I'm having a similar problem and in my inexpert opinion, I'm guessing it's this line in the log pointing to the problem

WARNING: accessing gdrive failed: <urlopen error [Errno -3] Temporary failure in name resolution>

[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/211082/motioneye.txt)

Calduino, 2016-04-09:
I started again from scratch with the SD card and it works fine now.

rushless, 2016-05-21:
Can confirm, reformatting the SD drive and reinstalling MotionEyeOs worked. Working with a Raspberry 3, had same problem as described in thread. Thanks!

Also, not sure if it matters, when I was on the new OS I started with google drive auth first, and got it working before doing anything else. I'll edit if I run into issues later in the process.

ggramlich, 2016-10-16:
I have a fresh installation of motioneyeos-raspberrypi3-20160919.img.
I configured the Google Drive as Upload Service and entered the autorization key. The Test service button  results in the message "Accessing the upload service succeeded!". (When revoking the access from the Google Dashboard, this has the expected effect.)
I never see any file uploaded to my Google Drive.
I tried with and without slashes at the beginning and end of the "Location", Include Subfolders ON and OFF.
I Enabled Debugging and looked into the log files, but I do not see any hint, that MotionEye tries to upload anything.
Locally, Pictures and Movies are stored, when motion is detected, and I can access them.

Is there anything that I can do to make the upload to Google Drive work?

bortek, 2017-03-01:
Is this still a problem ? Are you sure there is no other issues like you are behind a firewall or cannot access Gdrive for some other reason?

hansen-m, 2017-03-01:
I'm in a similar situation, where clicking on the "Test Service" button reports successful but no files are ever uploaded to Google Drive, although they are being saved locally. I've tried different combinations of the upload path without any luck.

rushless, 2017-03-01:
I reflashed my cards, changed my google password, somewhere along that
route it worked for me.

svoket, 2017-03-01:
-----

ggramlich, 2017-03-02:
I just updated through the browser interface to motionEye Version 20170212, Motion Version 4.0.1+git37b3595.
Still, the "Test Service" says "Accessing the upload service succeeded!", but it does not upload anything.

bortek, 2017-03-07:
OK then we need more info to dig this into. Can you take a look in the logs while it tries to copy over the files to Gdrive? What do you see there, any specific error message?

ramsesGIT, 2017-04-06:
Same issue. Confirms authorization key, however, nothing is uploaded. Please advise how I can help you resolve this issue. Thank you!

ccrisan, 2017-04-06:
motioneye.log file could give more details.

t6a, 2017-04-17:
My google drive upload stopped working - I think because I switched to Fast Network Camera Mode and later back. I had to switch file upload to google drive off, back on with a new google access key and then it worked again.

ggramlich, 2017-04-17:
I took a look at the motioneye.log again. I this is not related to the upload issue, since I get these entries regardless, whether I switched upload media on or off:
(I replaced the end of the signature string - as I don't know, if this should stay secret.)
```
2017-04-17 17:37:48: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=9dde09852175f492c403878b1c76a1xxxxxxxxx (127.0.0.1) 6.28ms
2017-04-17 17:37:48: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=9dde09852175f492c403878b1c76a1xxxxxxxxx (127.0.0.1) 3.97ms
2017-04-17 17:37:49: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=9dde09852175f492c403878b1c76a1xxxxxxxxx (127.0.0.1) 6.09ms
2017-04-17 17:38:18: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=f23609176a9fb1090af74b53ba1cf9cxxxxxxxxx (127.0.0.1) 3.39ms
2017-04-17 17:38:18: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=movie_end&thread_id=1&_signature=5cde6984922e80672327b405b09c26cxxxxxxxxx (127.0.0.1) 3.35ms
```

Apart from that, I do not see anything of interest logged (neither in the motion.log)

ccrisan, 2017-04-17:
@ggramlich do you have any special characters in your admin password? If yes, can you try simplifying it and see if the probme persists?

ggramlich, 2017-04-17:
@ccrisan Yes - there was a "special character" in the motioneye admin password.
I changed the password and now see the first images and videos uploaded to Google Drive.

Thanks a lot.

ccrisan, 2017-04-17:
@ggramlich can you tell me what that special character was so I can see if I can "add support" for it? Was it an ASCII char or some unicode?

ggramlich, 2017-04-17:
@ccrisan Giving away my secret sauce ;-)
ASCII 0x20

ccrisan, 2017-04-17:
Thanks :) I'll try to either make it work or validate the password to exclude unsupported characters.

closed 2017-06-01 by bortek

mfalcuta, 2017-08-10:
Hi ,
I cannot see File Storage option on a raspberry pi b+ 1.2.
Is this available only for newer boards?

keithmartinkinyua, 2020-03-25:
how could i save my footage from motion eye to aws s3

starbasessd, 2020-03-25:
Set up a Google drive, then
https://support.cloudhq.net/how-to-sync-amazon-s3-and-google-drive/

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

keithmartinkinyua, 2020-03-26:
it seems not to work.i'm having issues with s3 authentication

starbasessd, 2020-03-26:
Not a motionEye or motionEyeOS issue, then, if you can't get GDrive to talk
with AWS S3.
I have motionEye working with GDrive, no issues. Follow the step by step
instructions
when you click the 'Obtain Key' link.
After doing so, click "Apply' at the top BEFORE testing. It will fail,
forcing you to go through again the setup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?