motioneye-project/motioneyeos #230: EsCam QD300 with MotionEyeos

xxllo, 2016-03-24:


xxllo, 2016-03-24:
This translition is with Google
Have an IP Cam from ESCAM qd300 and can not do to activate the cam in Monioneye it . What am I doing wrong???

guadacorte, 2016-04-03:
i also have an ip camera. I cannot get it to work
i have a ideanext C1702W. Sometimes it will connect using rtsp but only a grey screen with the time and date stamp showing

xxllo, 2016-04-25:
look here
https://www.ispyconnect.com/man.aspx?n=ideanext

badele, 2016-10-01:
Hi,

I have received my new Escam Brick QD300. It work fine with motioneyeos. For that, i did it :
- Go to https://www.ispyconnect.com/man.aspx?n=eScam#
- Click to "Brick Qd300" and enter informations
- Go to motioneyeos and add network camera
- Paste the previous ispyconnect informations

Note: For configuring the camera (change résolution, network address, etc), it seem we must have a windows operating system.

For me this issue can be closed

ccrisan, 2016-10-01:
@badele thanks for sharing this with us. I'll close the issue then.

closed 2016-10-01 by ccrisan