motioneye-project/motioneyeos #232: Google drive - duplicate JPGs uploaded (tracking FPS setting)

edllew, 2016-03-25:
I have Still Images and Movies enabled, both motion triggered.  Am uploading to google drive.  This is essentially working for me.  But on google drive, duplicate JPG files are created, at the FPS rate.  Duplicates are not created on my local file system, only on google drive.

For example:  
At 10 FPS A 3 second event will create 1 AVI plus 3 JPG images locally.  On google drive I get 1 AVI and 30 JPG images.  Just 3 of those 30 JPGs are unique.  The rest are duplicates.  At 5 FPS, I get 1 AVI and 15 JPG images.

No clues for me in log files that I could detect.  I can send those along if it would help.

I am using an RPi 3, with the native camera, which is working well.  It was flaky at first because I was using a power supply without enough oomph.

ccrisan, 2016-03-26:
Given the default file name pattern for pictures, all pictures taken within a certain second will have the same name. On your disk they will be overwritten but on google drive they will be separate files sharing the same name. Don't save all frames - it doesn't make sense. Make a movie instead.

edllew, 2016-03-26:
Thanks for the response.  I don't really want to save all frames.  I'll try to explain what I want to accomplish:
-get an email with a still image or two showing the motion event.
-have access to a video on google drive showing the motion event.

Playing with the settings, it seems I need to have still images turned on in order to get images with the notification email.

digitalwaterfalls, 2016-03-28:
I noticed the same behaviour last night when I set up my RPI3 with build 20160320 and linked it to my Google Drive account. Files were being uploaded with the same filename (time as defaulted by the initial settings) so I added the date variable as a prefix to the filename as an experiment to try and differentiate between the files more, but the Google drive filenames still show the same names where sufficient frames were imaged, within a single second. I wonder if there is a millisecond variable that can be added to the output filename to help with making the filenames unique?

Just starting to get to grips with motioneyeos - fantastic system BTW!

bortek, 2017-03-07:
@edllew Had this been solved ?

closed 2017-06-01 by bortek