motioneye-project/motioneyeos #233: No Motion Detection Settings - Raspberry PI 3

dontspammexyz, 2016-03-26:
Hello, I get a brand new Raspberry Pi 3 with Pi Camera. MotionsEyeOS works fine. But  there are no motion detection settings - no "motion moving" tab. Live Picture works fine.

motion.log shows me:
`using motion binary /usr/bin/motion-mmal
[1976001408] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[1976001408] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[1976001408] [NTC] [ALL] motion_startup: Motion mmaltest Started
[1976001408] [NTC] [ALL] motion_startup: Logging to syslog
[1976001408] [NTC] [ALL] motion_startup: Using default log type (ALL)
[1976001408] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD`
[image]

How to bring it up to make motion detection/storing? THX

Gordon-1, 2016-03-26:
Try changing "Fast Network Camera" to OFF	in EXPERT settings

dontspammexyz, 2016-03-26:
This worked for me. Just I could add my local Raspberry Camera. Thank you!

closed 2016-03-26 by dontspammexyz