motioneye-project/motioneyeos #234: Working Schedule

fnordson, 2016-03-27:
Maybe I just thought about it too long but I have some trouble understanding the working schedule for my case.

I have the cam set to record outside of working schedule and set the schedule for the working days from 7:00 to 19:00. Now I would like to have the camera started watching from Friday 19:00 till 7:00 on Monday. But I'm not quite sure how to set the timers for Saturday and Sunday.

Currently I've set the days to 13:00 till 13:30 so there is a gab of 30mins at this time. 
I did that because I wasn't sure what happens if I set it to 0:00 till 0:00. 

Any ideas here?

digitizing, 2016-03-27:
From my experience this application does not understand the time across from mid-night to another day.  So each day needs to be defined for the application to behave the way you want it.  I hope this helps.

closed 2017-03-07 by bortek

anselal, 2019-04-24:
eg. I have set my schedule like this
[image]

Motion detection is enabled but when I step in front of the camera on a Monday-Friday before 08:00 the camera won't record anything.

Any help please ?

fnordson, 2019-04-24:
It's probably the issue I had originally: it won't go through  midnight. 
Set it to working OUTSIDE the schedule and set the schedule to 8:00 -> 17:00
For the weekend just chose some minute. For example: 13:37 -> 13:39

anselal, 2019-04-24:
@fnordson seems to work, thnx !!!

fnordson, 2019-04-24:
You're welcome @anselal :)