motioneye-project/motioneyeos #235: Motion detection makes camera stuttering

linuscool2, 2016-03-27:
Hello!

Just installed latest motioneyeos on my new RPi 2 with the CSI camera. 

But there is a big problem with the recording when motion detection activates. 
https://www.youtube.com/watch?v=E2UWvxSjUw4&feature=youtu.be

As you can see in the video it's starts to stuttering when the motion detections activates.
I have tried running it with 96, 128, 256 and 512MB GPU RAM. Various overclockings, lower resolution etc.
This worked a little better with the "Fast Network Camera" but this disable all the motion features that I need. 

The stuttering are still there if I watch the videos on my computer (downloaded from Google Drive). 

Does anyone know how to fix this?

Thank you!

cuddylier, 2016-03-27:
The exact same problem has been reported in #227 but for a RPI 3 instead so might be related, awaiting a reply on it.

perrychan, 2016-05-01:
This is also happening to me when recording to network share.

MMiknis, 2016-05-25:
Yep I also get the exact same issue on the RP1, 2 and 3. Once it detects motion and starts recording it drops FPS and starts to stutter

ccrisan, 2016-05-25:
Movie encoding is known to be a bottleneck so no wonder it behaves like that.

MMiknis, 2016-05-25:
Would be possible to use Pi2 as the capture device but get a Desktop machine do all the motion detection and recording?

weedy, 2016-09-02:
motioneye is threaded right?

ccrisan, 2016-09-02:
motionEye is not threaded (well, not in the way you'd think). Nevertheless, the entire video logic is implemented by the motion daemon, which is threaded, as far as I know.

closed 2017-03-07 by bortek