motioneye-project/motioneyeos #236: How to use area_detect

BerndFl, 2016-03-28:
Hi,

Currently the motion is triggered by any movement in the complete view of the camera. I need to limit this to a certain area, otherwise I get a lot of false hits. I added the line 'area_detect 456789' to the file /data/etc/thread-1.conf but that doesn't do the trick. What else do I need to do to get this working?

Thanks

Bernd

Gordon-1, 2016-03-29:
Until this feature is added to motioneyeos you need to manually  create a mask file and add it as an option in "Extra Motion Options" in the Video Options section.  Please see http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMaskFile and after reading that let us know if you need more guidance.

scohrab, 2016-04-29:
I'm puzzled by this answer, as I thought that the whole point of the 'Extra Motion Options' box was to be able to add in any valid Motion configuration option, presumably as defined by this list:-
http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions

So, I currently have the following lines in that box (latest MotionEyeOS running on RPi3 with a PiNoIR and one USB cam):-

smart_mask_speed 3
despeckle EeEedDdDl
area_detect 123456789

After applying that change (and I also rebooted, just to check that Motion had been restarted), my motion.log file looks like:-

using motion binary /usr/bin/motion-mrdave
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "despeckle"
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "despeckle"
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

I'm not really sure what this means. Is it just that despeckle is not recognised by the version of motion (Mr Dave), and the other two parameters (area_detect, and smart_mask_speed) are ?

Or does Crisan have to explicitly enable MotionEyeOS to recognise config options like these ?

As an additional query, does anyone know where to find full config option documentation for this specific version of Mr Dave's Motion, or are they the same as the main ones ?

ccrisan, 2016-04-29:
Nope, I don't have to enable anything. It's just that motion config options have changed over time and currently motionEye knows two dialects (called "old" and "new"). Mr. Dave's motion speaks the "new" dialect. Every option controlled by the UI is translated to the correct dialect (if available) and therefore the dialect differences are transparent for the user, regardless of the version of motion used.

Now, for the extra motion options, you are almost correct. You could add there any option supported by your motion version, provided it's not already covered by the UI. The mask-related options are not covered by the UI, so you should be able to use them.

The despecke option is not covered by the UI either, but it seems that it's not supported by the new dialect. Looking through Mr. Dave's [source code](https://github.com/Mr-Dave/motion/blob/master/conf.c#L553), there is this `despeckle_filter` option that may be the new dialect version of the old `despeckle` option.

I hope things are clear(er) now.

scohrab, 2016-04-29:
Yes, much clearer now, and that reference to Mr Dave's source code also seems to confirm that both area_detect and smart_mask speed should work OK, as they both appear in that area of the source code. I have changed my config, and now there is no error in motion.log, so that also seems to confirm that all is OK.

Thanks very much for the quick reply, and also for the really great package.

closed 2017-03-07 by bortek