motioneye-project/motioneyeos #239: Motion trigger log file?

CrazyEyesPete, 2016-03-30:
Is there a log file that indicates or counts the number of times a motion event was triggered?

I have an idea to use this as a people traffic count system for digital signs, which are also Pi based. 

Using an area mask, I want to count the number of people that stop directly in front of my televisions to look at the digital signage.

I have the motion detection settings adjusted quite well to detect when a person comes in front of the camera and stays for a while and then leaves. But I'd also like to count the number of times this type of event happens.

Maybe there is already a log file created and I'm just not finding it?

I just started using MotionEyeOS 24 hours ago and I'm already blow away by its features! I'm currently setting it up in my office for security, but wondered if I could also us it for market research by counting people traffic.

ccrisan, 2016-03-31:
There's no such thing out-of-the-box but with a small piece of code you could count these events, using the *Run A Command* option from the *Motion Notifications* settings section.

CrazyEyesPete, 2016-03-31:
Will look into that. Thank you.

closed 2017-03-07 by bortek