motioneye-project/motioneyeos #242: WPA Supplicant Fail with driver name in front

joubin, 2016-04-02:
At boot, Im getting 

```
 * Starting wpa_supplicant: failed (r8712u)
 * Configuring wireless network: no link
```

Could this be a driver issue, or could it be my newbie ness using wpa_supplicant?

ccrisan, 2016-04-03:
When that shows up, it's usually one of the following:

* your network encryption type is not WPA(2)
* your network SSID or PSK don't match
* your network SSID or PSK have special characters in them (this shouldn't normally pose problems but you never know)
* the wifi signal is relatively low in that area
* your router is malfunctioning

jonfairbanks, 2016-07-23:
I am also getting this error with multiple wifi dongles (r8188eu & 8192cu) on a Pi Zero v1.3 and am running out of things to try. 

To start:
1. Flash motioneyeos-raspberrypi-20160605 to sd card in Windows
2. Copy [properly formatted](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-%28Plus%29-And-Zero) wpa_supplicant.conf to root of sd card
3. Get "Starting wpa_supplicant: failed (r8188eu)" on boot
3a. If the driver name is printed, I'd assume loading of the driver was successful. 
4. Enter boot loop due to no internet connection

Both of these wifi dongles have worked out of the box when switching to a Raspian install. It should also be noted that I have a Rpi3 running motioneye on the same wireless network with no issues, so the AP seems fine. 

Things I've checked:
- Two Pi Zero v1.3's
- Camera/No Camera
- Two different dongles with Raspbian support (r8188eu & 8192cu)
- I am no more than 1 foot from my wireless router
- Although Rpi3 works on the same network, switched network from WPA2-PSK[AES] to WPA-PSK [TKIP] + WPA2-PSK [AES] and back again.
- Two power supplies (2A/2.5A) and 3 power cords
- wpa_supplicant.conf saved with correct extension and no extra characters or spaces
- Triple checked ssid and psk and copied straight from router's config
- Nothing weird in wireless network name or password

Any thoughts on what may be going wrong or what else I can check? Is there a more verbose log somewhere?

ccrisan, 2016-07-23:
That's odd. Could you please try it with 20160705?

jonfairbanks, 2016-07-27:
Thanks for the response. Things seem to be working with 20160705. :+1: 

Can [this page](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices) be updated to reflect the latest version?

ccrisan, 2016-07-27:
Yes, as soon as the release is marked as stable.

closed 2016-07-27 by ccrisan

FintaruS, 2018-08-06:
just had this problem, you can not uses a "}" in the password.... making a new password worked for me :)

senojed, 2018-12-24:
got this problem with latest release 20180627. Also tried 20181209. But mentioned 20160705 is working ok...
setup: PI Zero 1.3 + TL-WN725N

sarathchandars, 2020-04-26:
got this problem today with 20190911 release on my Pi Zero (not W) with a TL-WN725N: Error message: Starting wpa_supplicant: failed (r8818eu).

ashu15645, 2020-07-12:
@sarathchandars were you able to get around this issue ? I am getting the same issue with my Rpi1 ( Error Message : wpa_supplicant:failed (r8712u))