motioneye-project/motioneyeos #243: don't get any settings options with new setup

pumbaa2183, 2016-04-03:
I decided to build a 2nd, camera but after installing and setup the OS on my SD card and logging into the web interface I discovered that I do not have the options to setup anything for the camera ( please see screen shots) 

I have formatted and installed different versions of the OS several times ..... Suggestions ?
[image]

ccrisan, 2016-04-03:
Close the menu, use the key icon to log in as administrator and do your stuff.

closed 2016-04-03 by ccrisan

pumbaa2183, 2016-04-04:
Thats the screen I get after first boot up before I can log in as admin, when I am logged in as the default admin.

ccrisan, 2016-04-04:
Does the settings menu automatically open after first boot?

pumbaa2183, 2016-04-04:
yes it does, but it opens to what you see in the screen shot --- so I don't get the option to set an admin username and password, or set my Wifi info :( 

Sent from Alto - altomail.com

ccrisan, 2016-04-04:
If it does, close it and use the key icon to log in as administrator.