motioneye-project/motioneyeos #244: Email Sent Without Attachment

EricFries, 2016-04-04:
I'm experiencing an issue where some motion notification emails don't have any image attachments.  I'm using a Raspberry Pi 3 with 20160320.

For example, a 4:14 PM email was sent without an attachment, and the corresponding log entry in motioneye.log doesn't indicate an error:

`2016-04-04 16:14:43: [sendmail]     INFO: email sent`

Most emails send fine with image attachments.

ccrisan, 2016-04-05:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails

EricFries, 2016-04-06:
Great, thanks, sorry for not seeing that first!

EricFries, 2016-04-06:
I did notice this error in there also:  

`2016-04-05 20:44:56: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081`

Not sure if it's related or a separate issue.

ccrisan, 2016-04-06:
Not related at all. It's probably due to the motion daemon sometimes not being as responsive as it should.

EricFries, 2016-04-08:
Cool, it's fixed.

closed 2016-04-08 by EricFries

glctesta, 2016-11-26:
Hi, 
I've checked all the parameters which you was speaking about, but emails are sent without attachment too. On top on this, I am not able to activate the google drive store procedure. If I click test, it's successful, but no files saved on my google drive, can you support me?

gusv, 2017-11-18:
Hi. I installed ME as OS on a PI3 and it worked perfect, motion, email, attachments, etc. I re-installed on Raspbian following your instructions and also worked fine. The only thing is that I cannot get the motion to work. The test email is fine but there are not motion detected and no email sent even though I used the same configuration for still images, etc. Am I missing something?

surak, 2019-05-07:
From the faq:

> Just enable still images and set Capture Mode to Motion Triggered. Also make sure to configure _Attached Pictures Time Span_ to a reasonable amount of seconds (5 to 30).

This is not a setting in either _Still images_, _Motion Detection_ nor _Motion notifications_.

moha-klassic, 2019-05-07:
hello. I'm experiencing an issue where some motion notification emails don't have any image attachments. I'm using a Raspberry Pi 3 with motionEyeOS 20190119.

This is what i received in my mail.
Motion has been detected by camera "Camera1/meye-f69e7e8a" at 2019-05-07 20:54:09 (Africa/Accra).