motioneye-project/motioneyeos #245: LED Control (not CSI)

LackingAce, 2016-04-06:
So i was looking through the settings wondering what to do... then i saw under Motion Notification, "Run a Command"! The Idea hit me, can i set it up to turn a led on whenever motion is detected?! Here's where you come in... i dont know how to. As in, im not sure what code to use. Also, when i did try a code, i went to go paste it... then found out there is only one line to type in a command. So if i want to put in multiple lines to turn the led on... i cant. I will List to you what im using in this setup. 
Materials:
-Raspberry Pi 2 Model B
-HD IR Pi Camera
-High Flux White LED w/ Resistor

[image]
[image]

sora1984, 2016-04-20:
Couldn't you try something like this?

```python
import time
import RPi.GPIO as GPIO ## Import GPIO library
GPIO.setmode(GPIO.BOARD) ## Use board pin numbering
GPIO.setup(7, GPIO.OUT) ## Setup GPIO Pin 7 to OUT
GPIO.output(7,True) ## Turn on GPIO pin 7
time.sleep(5) ## Delays for 5 seconds
GPIO.output(7,False) ## Turn off GPIO pin 7
GPIO.cleanup() 
`

LackingAce, 2016-04-20:
if your saying to try that under the "Run a Command" option... then i cant, like i stated above. It only allows for one line of code.

sora1984, 2016-04-21:
I should have said make it a executable script and call it from the "Run a Command" option.
Using the terminal run these commands one by one.

_wget https://dl.dropboxusercontent.com/u/19918926/python/led.py
chmod +x led.py
cp led.py /sbin_

You can now just type _led.py_ and the script will run.
Enter _led.py_ into the "Run a Command" box and save.

LackingAce, 2016-04-21:
i should have thought of that... I will try it out and get back to you. Thanks!

LackingAce, 2016-04-21:
i think i've run into a snag... it connects to the file... put tells me its a "Read-only file system". The next line says "Cannot write to 'led.py' (Success)". When i try "chmod +x led.py" it tells me that there is no such file or directory. Did i do something wrong? is there a way i can just make that file in the console? then run the 2 other commands you told me to run?

sora1984, 2016-04-21:
Try it this way around

_mount -o rw,remount /sbin 
cd /sbin
wget https://dl.dropboxusercontent.com/u/19918926/python/led.py
chmod +x led.py_

then run the script see if it works
just connect the led + to pin 7 and - to pin 3 

_http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/09/Raspberry-Pi-GPIO-Layout-Revision-1.png_

LackingAce, 2016-04-21:
Still getting the same thing. When i try "mount -o rw,remount /sbin" i get "cannot find /sbin in /proc/mounts" Like i said, is there a text editor of some sort that i can use to just type the code myself? Just to let you know i did realized a bit ago i was using an older version but did just upgrade with a fresh new install... and it still didn't work.

sora1984, 2016-04-21:
mount -o rw,remount / 
nano is a text editor but if it's read only then you will still get denied. 

Best bet is to install the script to the root folder and change the line of "run a command" to /root/./led.py

LackingAce, 2016-04-21:
ok so i tried to at least run the code through the python3 command and it worked, in Raspbian! I did have to remove the two "print" commands. I did run into another snag, i have a problem where when ever i install motioneyeos through raspbian, if i turn off the rpi, and turn it back on, it wont turn on... as in, it just sits there with a flashing "_" in the top left. For what ever reason, i must of screwed something up this time, before ever installing motioneye, because it did the flashing "_" when i went to restart it after checking to see if the code would run with python (In Raspbian). So now i have to go and reinstall Raspbian... again. Otherwise, your code and everything should work... just it has to be the Raspbian version. It wont work on the version where you write a motioneye.img to a sd card.

closed 2017-03-07 by bortek