motioneye-project/motioneyeos #247: Gray images with "unable to open video device" - could be related to FPS drop on motion detection

ShacharWeis, 2016-04-09:
Raspberry Pi version 2, two Logitech webcams, GPU memory at 256, all other settings are default. Motion causes FPS to drop to 0.1. After 12 hours the dropbox folder is full of gray images with "unable to open video device". This happens also when there is no motion, and causes a false positive motion trigger. Sadly, this renders the system unusable.

ccrisan, 2016-04-09:
Many users have reported issues with Logitech webcams, especially when using multiple such devices. At this point I'd recommend trying with a different camera model (or at least two different camera models at the same time).

xxllo, 2016-04-15:
Have the same message with a network camera of the brand ESCAM Brick QD300

pwlkania, 2016-08-02:
Have the same issue with IMX219 (https://www.raspberrypi.org/blog/new-8-megapixel-camera-board-sale-25/).
Any advice?
Thanks

elstimate, 2017-01-04:
hello.

same camera, same problem.
technically, ive tried tremendous amont of combination in rtsp, http, URL-s in motion eye.
when I try to use http, ME says "not supported".
when I try with rtsp protocol, camera appears as TCP, UDP style... none of them gives live image
i was trying with user name and password and channel and stream injected in the URL line... no success..

this is the 3rd night in a row i "play" with this... quite frustrating.... any help to solve?

elstimate, 2017-01-04:
now, im using wireshark with winpcap in order to catch the network communication when using VLC network stream play... no success.

what is sure: in http, with the factory built-in webpage, camera can provide video strem through browser.
so, camera is working, video stream via http is valid

ccrisan, 2017-01-04:
Wireshark is actually a very good idea to debug this kind of problems.

If when you use an HTTP URL it complains about a non-supported camera, then you must have typed the wrong URL. Don't just use the webpage (web UI) URL of your camera; you'll need to search for the MJPG URL, which, when visited directly on your browser, should display *a standalone MJPG image* without any surrounding webpage interface elements.

If you have problems with the RTSP URL, please report the problem to [the motion project](https://github.com/Motion-Project/motion), including motion logs and configuration files (w/o sensitive data).

elstimate, 2017-01-04:
Hello  Crisan,

THanks for your help!

May you give me an example URL how this MJPG URL looks usually like?

This ESCAM camera needs authentication, what I trying to use about RTSP is
something like:

 rtsp://192.168.1.10/user=User&password=Password&channel=1&stream=1.sdp

Coudl you give me a HTTP like URL example?

In wireshark, I have not found a particular moment, when a camera sends a
HTTP protocol with the video stream link inside.
I could catch the connection packages, ACK and other handshakes, but when
HTTP streams goes, those are just videos stream packets, without HTTP URL
links   (as source link).

Thanks!
Zsolt

elstimate, 2017-01-04:
Hello,

I'm affraid, HTTP based video stream is a dead end... (or I cannot see
further)

Since there is an authentication method at the beginning of the factory
built-in web logon, motion eye cannot perform the very same authentication
 (or I'm wrong)

but here is what I captured with wire shark when I did the factory site log
on  (authentication) and than I got the video stream via the web broswser,
using the built-in site of the camera:

[Full request URI:
http://192.168.1.106/web/cgi-bin/hi3510/param.cgi?cmd=getvencattr&-chn=011]
[Full request URI:
http://192.168.1.106/web/cgi-bin/hi3510/getaencattr.cgi?-chn=011]

Request URI:
http://192.168.1.106:80/livestream/11?action=play&media=video_audio_data
Request URI Path: http://192.168.1.106:80/livestream/11

not only these, but these were important.
so, #1) client makes auth. with camera  using user name and password,
than camera checks the credentila, and if valid, will provide the video
stream for the client (browser)

I just assume how motion eye works with the credentials, but my you can
clarify the how the user name and password works in ME.... is it an URL
overwrite or does a "hand shake" with the camera via HTTP protocol or so?

error message when I was trying with links above:

connection timed out

THanks!
Zsolt

2017-01-04 15:16 GMT+01:00 Zsolt Szentjóbi <elstimate@gmail.com>:

bortek, 2017-03-07:
I don't think there is more to add here. You can start a discussion in the project's google group if you want to get feedback from other users. thanks.

closed 2017-03-07 by bortek