motioneye-project/motioneyeos #248: direct video stream url

aristosv, 2016-04-09:
I have a raspberry pi 3 with the raspberry pi camera v1.3 connected on it, running the latest version of motioneyeos.

I am trying to connect the pi/camera with a third party program (webcamxp, sighthound etc) but I don't know the URL of the video stream. I tried scanning for it using ispy software, but it can't find anything.

What is the direct video stream url when the pi is running motioneyeos?

Thanks

ccrisan, 2016-04-10:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#whats-the-deal-with-snapshot-url-streaming-url-and-embed-url

closed 2016-04-10 by ccrisan

cederlof, 2017-09-12:
@ccrisan There is only a description on provided URL, no example of a streaming URL?

ccrisan, 2017-09-12:
@cederlof if you click on it, a popup window should appear with the actual link to copy.

Bluscream, 2017-12-08:
what about RTMP?

IngiDingi, 2019-08-30:
Hi.

I also dont know how to build the right URL for streaming.
I followed the LINK which is mentiond above but sorry, I dont get it.
Can you provide an example of an URL so I can understand how to use it?

Thanks
Ingo