motioneye-project/motioneyeos #249: No Camera When Using Network Share

tuanndryan, 2016-04-12:
Hello everyone,

I'm using Pi 2 with Pi Camera module. Every time I use network share, the camera is OFF. I'm setting my network share as:
Network server: ENVY15 (name of my laptop)
Share Name: piCAM (video is supposed to be saved in this folder)
When I changed back to custom path, the camera works well. Dropbox uploading works normally. Any ideas to solve this problem?

Thanks

foresthus, 2016-04-13:
Hi,

I have got the same problem. I hope nfs will be soon available.

cu

davisstu, 2016-04-25:
The camera doesn't work if you enter your network share credentials incorrectly - I suspect that this is what is happening here. There is a more detailed guide to this in the wiki.

veames, 2016-07-11:
This happened with the older motionpi software. If you lose connection to your smb share, the camera is blank or greyed-out. Previously I had fixed this with a watchdog timer that restarts the PI if it's unable to ping the samba/windows server.

ccrisan, 2016-07-11:
The new prerelease has a test button to this end. Have you tried it?

closed 2017-06-01 by bortek