motioneye-project/motioneyeos #250: Testing a BananaPi

Chiny91, 2016-04-12:
I've semi given up on running motionEyeOS on a Raspberry Pi (reasons elsewhere) and shifted over to a Banana Pi.   From opening my front door to the Amazon delivery man to having all 4 IP cams up/running was just under 15 minutes, a real tribute to motionEyeOS as a straightforward system - many thanks.

It is far too early to assess the results yet but I am hoping the faster ethernet will make the difference.   Interestingly, the cpu oomph does not; the R Pi 2 reports 25% usage whereas the Banana Pi reports 50%.

ccrisan, 2016-04-12:
Well I guess it's kind of normal that the Banana PI with 2 cores reports twice as much CPU usage as the RPI 2 with 4 cores. Other than that, I also find the Banana PI a great board.

Chiny91, 2016-04-12:
dmesg appears to be filling up with approx 2 lines/second of:
_[Thu Jan  1 01:04:45 1970] clocksource hstimer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 21718868714 ns_

Despite that dodgy timestamp, the clock is set:
_# date
Tue Apr 12 20:40:13 BST 2016
[root@bananapi ~]#_

I'd better poke about through the various logs.

ccrisan, 2016-04-12:
Hmm that's odd. I'll try to reproduce this myself.

Chiny91, 2016-04-13:
I've stopped dmesg filling up; the cause was a test camera with a faulty config.   The fault was having Dropbox configured but not authorised.   I'm not sure that dmesg should have been affected but the problem has gone away.

12 hours testing... the Banana Pi has not been frequently rebooting as did the Raspberry Pi and has a much more responsive (quicker) web page.   The next test is to try and break it all by getting it to backup to DropBox :smirk:

closed 2017-06-01 by bortek