motioneye-project/motioneyeos #251: ftp to sd card

Tntceria, 2016-04-12:
When I ftp to  motioneye and look in the sdcard directory I have no files.  I do have file storage on for the sdcard and I have ftp enabled.  When I look in the /data directory I do see the camera movies and pictures.   
  When I turn file storage on for a usb device and ftp to motioneye the files do appear under storage/sda1 as expected. 
  It appears there is a problem with the mount bind does anyone else have this problem??

ccrisan, 2016-04-17:
What motionEyeOS version do you use? Do your *sdcard* files show up in `/data/output` (where the ftp sdcard mount should point to)?

Tntceria, 2016-04-18:
I am using version 20160320, the files are showing up in /data/ , /data/output/  is empty so I guess that is the problem. Is there away to change the storage to go to /data/output/ ?

ccrisan, 2016-04-18:
Of course, you should simply change the *Root Directory* setting accordingly. It is set to `/data/output` by default but you have probably changed it afterwards.

closed 2016-04-20 by Tntceria