motioneye-project/motioneyeos #252: Orange PI PC support

sambul13, 2016-04-13:
Hi,

Can you add [Orange PI PC](https://linux-sunxi.org/Orange_Pi_PC) to supported devices?

ccrisan, 2016-04-13:
I don't own such a device yet :)

sambul13, 2016-04-13:
They're sold for $15 [here](http://www.aliexpress.com/item/Orange-Pi-PC-linux-and-android-mini-PC-Beyond-Raspberry-Pi-2/32448079125.html), hugely popular, and already have several working large Linux distributions [adapted](http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=342). Unfortunately, none of these distros is lightweight and specialized in motion detection & surveillance, while this device seems to ideally suit that purpose in home use. Please consider. :)

Chiny91, 2016-04-14:
Maybe a donation of one such device ordered/delivered to the developer's selected address would encourage this bespoke software :expressionless:

sambul13, 2016-04-14:
Contacting the Ali seller at the above link who is also owner of the company making the device might result in getting a few different type Orange devices by the developer, since demand for home & small office surveillance devices running on ARM is steadily growing among users.

soroushcrazy1, 2016-04-15:
I have "orange pi one" witch has 4core 1.2Ghz Arm and 512MB Ram on it.it has Ethernet and 2 Usb (one micro and one usb2) it has camera port and power with 5volt.
it price is 9$. I think this is beter than raspberry pi Zero (because has Ethernet port)
if anyone cam Work on it , it would be perfect.
site link:
http://www.orangepi.org/

Roger53, 2016-06-03:
Hi ccrisan, great software on top of a great motion package !

OS version for Orange Pi One (OPiO) : Yes please, gives me back GB's on the SD card for a picture store? Pretty powerful SBC.

Currently I install it on top of Armbian 3.4. Works fine, just minor camera issues in the log as Orange Pi camera on Armbian is still under development. still pictures / movie / motion detection / mailing etc work fine.

OPiO and a 2MP Orange Pi One Camera = $15 + shipping cost, if bought as a set.
Camera gives a bit of a red-ish low contrast picture, so I want to correct for that, and somehow that only works in the OS version, or should I check the option list ?

OPiO is also much smaller then all the rest, so fits easier in dummy camera housing.

gtr33m, 2016-10-05:
Motioneye (not os)  works very well with Orange Pi PC when installed on top of Armbian 3.4

Installed using the Raspbian instructions, including pre-built ffmpeg package. Only issue was python pytz was missing, but install able with pip.

Great for multiple network cameras as they are cheap and can run 2 cameras per SBC comfortably.

ccrisan, 2017-01-01:
See https://github.com/ccrisan/motioneyeos/issues/649.

ericjanvanputten, 2017-03-03:
@gtr33m May i ask how you installed it. 
Based on another community member i got 2 orange pi zero's and i cant get the web interface to load.
Not sure where i go wrong as a noob. I followed the raspbian instruction, but to bad no sigar yet.

gtr33m, 2017-03-03:
It was a while ago, but as I remember the only issue I really had was in
installing ffmpeg. Otherwise it was the raspbian instructions on armbian.
Did you get armbian up and going properly before installing motioneye?

ericjanvanputten, 2017-03-03:
Very probably not, I’m to much of a noob.
Copy pasting every instruction and for the rest google ;)

I tried Armbian, Ubuntu and debian, all no success yet.

Eric Jan C. van Putten

sr. Marketing Operation Center Manager
Email: ejvp@sitecore.net<mailto:ejvp@sitecore.net>
Mobile: +31 (0) 6 468 32 447

[sitecore own the experience small]

*Mark Hawkins*

This message and its attachments are for the designated recipient only and may contain proprietary or otherwise confidential information. If you have received this message in error, please notify the sender immediately and delete the original. Any other sharing, copying or use of this message by you is strictly prohibited.

ericjanvanputten, 2017-03-03:
I actually came a small step in the right direction - seemed the motioneye part installing kept going wrong, attempt 86 helped...

closed 2017-06-01 by bortek