motioneye-project/motioneyeos #255: URGENT! Latest update brakes the notification settings!!

kodapaio, 2016-04-15:
[image]

After the latest release with the support for setting your own hostname, everything seems to have been pushed one row down.

Could you please look over this and apply a fix for it? :) Both my cameras ended up liked this...

ccrisan, 2016-04-15:
What was the previous version that you've been running? DId you "skip" a version when updating?

kodapaio, 2016-04-16:
Nope, the latest version was the one from jan that I was running. And updated to the latest.

Mikiya83, 2016-04-16:
Same problem here, without "skipping" version (i'm on "release" updates only)

ccrisan, 2016-04-17:
I admit I added an extra parameter to the SMTP settings (namely *From Address*) but I wrote the code in such a manner that it would detect old vs. new config lines. I'm afraid that in your case you'll need to reconfigure your SMTP settings from scratch. I'm sorry for the inconvenience and I promise not to add any more SMTP params in the near future :)

Mikiya83, 2016-04-17:
Oh no prob man, it takes 10s to reconfigure SMTP server ;) So feel free to timprove your amazing project when you want, thanks for your work ! We signal the problem but it's not a big deal :+1:

closed 2017-06-01 by bortek