motioneye-project/motioneyeos #256: endless reboot on startup - not sure if wifi dongle not recognized

davecampbell, 2016-04-19:
apologies if i haven't taken steps already outlined somewhere - i tried to review at least the first 5 pages of reported issues.  and i didn't find an 'expert settings' section in the wiki, despite a reference to it.

fairly noobish here - but i was able to write a disk image for my raspberry pi 2 using diskimagewriter, and inserted into my raspberry pi 2, which has a wifi dongle attached.

i did not see the raspberry pi show up on the wifi network, so i hooked up an HDMI display and a usb keyboard.  all i see is the continuous reboot.

i guess i'm wondering how to break out of the continuous reboot in order to setup the network correctly.

i'm a bit concerned that the boot sequence only mentions a 'wired network' as if it does not recognize my wifi dongle.

hope this is pretty basic initial configuration stuff - i can deal with most linux cli things, so look forward to some advice.

davecampbell, 2016-04-19:
[image]
[image]

ccrisan, 2016-04-19:
If the boot log doesn't say anything at all about a wifi connection, then you probably haven't configured it. In order to enable wifi, you should first connect it to a DHCP-enabled wired network and access the UI using its wired IP address.

davecampbell, 2016-04-19:
could not access the boot log since it was cycling continuously, but connecting it to a wired connection allowed it to start normally.

then, of course, everything went swimmingly.

i'm not sure the installation instructions made that very clear though.  i hadn't connected my raspberry pi to a wired connection for a long time, so it wasn't my default response to a problem.

maybe the boot routine could recognize an initial setup and provide just a one-line suggestion when it detects no network on the wired port.  and/or a special call-out for this condition in the installation instructions.

thanks again - awesome project.

hxpaul, 2016-04-20:
This is where I got stuck too. The wiki says I should be able to preconfigure my wireless network with `./writeimage.sh ...  -n "myssid:mykey"` but it hasn't worked for me. I loved the idea of this. Will have to try later when I can get a screen near the router, generally never connected the rasperry pi's via ethernet.

ccrisan, 2016-04-20:
I'd need someone experienced with OSX to help debugging this problem. Normally the script will try to mount the boot partition (fat32) and create a wpa_supplicant.conf on it. I have no idea why the mount fails like that on OSX.

bortek, 2017-06-01:
closing this due to inactivity.

closed 2017-06-01 by bortek