motioneye-project/motioneyeos #257: corrupt files

Tntceria, 2016-04-19:
I noticed as the file storage starts to get large 100+ files. They also become corrupt so that the files cannot be opened.  Could this problem be related to the 10 sec file listing timeout.

ccrisan, 2016-04-19:
The corruption normally has nothing to do with the number of files. It might be that the more files you have, the bigger the probability is that a file gets corrupted.

Tntceria, 2016-04-19:
Once the problem starts occurring it seems to happen to all falls that are stored.  I also tried saving to a thumb drive and the results seem to be the same.

ccrisan, 2016-04-19:
How do you know the files are corrupted?

Tntceria, 2016-04-19:
When I try to view them i get a message that states part of the file is either missing or corrupt

ccrisan, 2016-04-19:
Can you attach such a file here?

Tntceria, 2016-04-19:
Ok here is one. 
[image]

closed 2016-04-20 by Tntceria