motioneye-project/motioneyeos #260: Fails to display video or any kind of image.

jagd1000, 2016-04-23:
Hi All;

After "burning" the sd card, and start the rapberry pi with the motioneyeos, I have to say I am said, no image at all.
I could not find any information how to debug the motioneyeos, All the debuig files I am aware attached.

Thanks to all Who works on the project. If I can help?

HW: Raspberry pi B; Camera Raspberry Pi Camera Module.

It is a good practice to have a debug hands-on like:
1. Verify the HW (camera)
......Commands /outputs
2. Verify drivers
...........
3. verify motion
..........
4. verify motioneyeos
...........

---Logs ---
[root@meye-9a325050 log]# cat motion.log
using motion binary /usr/bin/motion-mmal
[-1246108792] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[-1246108792] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[-1246108792] [NTC] [ALL] motion_startup: Motion mmaltest Started
[-1246108792] [NTC] [ALL] motion_startup: Logging to syslog
[-1246108792] [NTC] [ALL] motion_startup: Using default log type (ALL)
[-1246108792] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

==========================v42ctl info========================

[root@meye-9a325050 log]# v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : bm2835 mmal
        Card type     : mmal service 16.1
        Bus info      : platform:bcm2835-v4l2
        Driver version: 4.1.18
        Capabilities  : 0x85200005
                Video Capture
                Video Overlay
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x05200005
                Video Capture
                Video Overlay
                Read/Write
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 0: ok)
Format Video Capture:
        Width/Height      : 800/480
        Pixel Format      : 'YU12'
        Field             : None
        Bytes per Line    : 800
        Size Image        : 576000
        Colorspace        : SMPTE 170M
        Transfer Function : Default
        YCbCr Encoding    : Default
        Quantization      : Default
        Flags             :
Format Video Overlay:
        Left/Top    : 150/50
        Width/Height: 1024/768
        Field       : None
        Chroma Key  : 0x00000000
        Global Alpha: 0x00
[... 818 lines cut ...]
[Thu Jan  1 00:00:04 1970] pcm512x 1-004d: Failed to reset device: -5
[Thu Jan  1 00:00:04 1970] pcm512x: probe of 1-004d failed with error -5
[Thu Jan  1 00:00:04 1970] pcm512x 1-004c: Failed to reset device: -5
[Thu Jan  1 00:00:04 1970] pcm512x: probe of 1-004c failed with error -5
[Thu Jan  1 00:00:04 1970] wm8804 1-003b: Failed to read device ID: -5
[Thu Jan  1 00:00:04 1970] wm8804: probe of 1-003b failed with error -5
[Thu Jan  1 00:00:06 1970] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Thu Jan  1 00:00:07 1970] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[Thu Jan  1 00:00:10 1970] system date is 2016-04-23 09:29:47
[Thu Jan  1 00:01:55 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:03:28 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:05:00 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:06:33 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:08:05 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:09:38 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:11:11 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:12:43 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:14:16 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:15:48 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:17:21 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:18:53 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:20:26 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:21:43 1970] bcm2835-v4l2: error 0 waiting for frame completion
[Thu Jan  1 00:23:15 1970] bcm2835-v4l2: error 0 waiting for frame completion
[root@meye-9a325050 log]#

ccrisan, 2016-04-23:
It would be a lot easier if you just downloaded the log files from the Expert Settings section, in the UI and attached them here.

jagd1000, 2016-04-23:
Sorry, I am new in GitHub; I tried to attach files without success. I can do that. Could you drop the link on how to attach files?
Drag/drop doesn't work, I don't see expert settings option ?

Thanks,
JD

ccrisan, 2016-04-23:
You have a small remark at the bottom of the entry you're typing the comment in. It says *Attach files by dragging & dropping, selecting them, or pasting from the clipboard*.

jagd1000, 2016-04-23:
Ok,

Files attached;
Waiting for sugestion to advance,
I am performing the same setup with raspbian.

[v4l2-ctll-all.txt](https://github.com/ccrisan/motioneyeos/files/232985/v4l2-ctll-all.txt)

[motioneye-no-image.zip](https://github.com/ccrisan/motioneyeos/files/232989/motioneye-no-image.zip)

JD

ccrisan, 2016-04-23:
Is this a clean install?

jagd1000, 2016-04-23:
Yes;
Burn image some minutes ago.
The camera have the Led On;
I will try now raspbian to check the HW.

ccrisan, 2016-04-23:
Some log files are incomplete. At least the `dmesg.log` file is definitely not the entire file. You seem to have "downloaded" the files using `cat` and `tail`.

jagd1000, 2016-04-23:
Now, download using motioneye web page.

[motioneye-no-image-001.zip](https://github.com/ccrisan/motioneyeos/files/233002/motioneye-no-image-001.zip)

ccrisan, 2016-04-23:
Everything looks alright in your log files. Let's see if this works with Raspbian.

jagd1000, 2016-04-23:
in motioneye.log you have a connection fail every second. That doesn't seem normal to me.
Assertions:
file dmesg.log:
[Thu Jan  1 00:02:17 1970] bcm2835-v4l2: error 0 waiting for frame completion
file motion.log:
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
file motioneye.log:
2016-04-23 14:34:20: [motioneye]  WARNING: Connect error on fd 13: ECONNREFUSED
2016-04-23 14:34:21: [motioneye]  WARNING: Connect error on fd 13: ECONNREFUSED

file 
Apr 23 14:50:35 meye-9a325050 user.err motion-mmal: [0] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
Apr 23 14:51:35 meye-9a325050 user.err motion-mmal: [0] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful,killing it!
Apr 23 14:51:38 meye-9a325050 user.warn motion-mmal: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Apr 23 14:52:04 meye-9a325050 user.notice motion-mmal: [-1246157944] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Apr 23 14:52:04 meye-9a325050 user.notice motion-mmal: [-1246157944] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Apr 23 14:52:04 meye-9a325050 user.notice motion-mmal: [-1246157944] [NTC] [ALL] motion_startup: Motion mmaltest Started
Apr 23 14:52:04 meye-9a325050 user.notice motion-mmal: [-1246157944] [NTC] [ALL] motion_startup: Logging to syslog
Apr 23 14:52:04 meye-9a325050 user.notice motion-mmal: [-1246157944] [NTC] [ALL] motion_startup: Using default log type (ALL)
Apr 23 14:52:04 meye-9a325050 user.notice motion-mmal: [-1246157944] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Apr 23 14:52:04 meye-9a325050 user.warn motion-mmal: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Apr 23 14:52:34 meye-9a325050 user.err motion-mmal: [0] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart

ccrisan, 2016-04-23:
That's consistent with what you're experiencing, indeed. However the rest of the log files (where I would have expected some hardware-related issues) everything looks clear.

jagd1000, 2016-04-23:
Ok, good, I will be checking Everything with Raspbien, I will post the results,
Burning image now ..

jagd1000, 2016-04-23:
Hi All,

Checked with Raspbian; camera not detected, 
After disconnect and connect all cables; it is working.
Ticket Closed.
Thanks,
JD

closed 2016-04-23 by jagd1000