motioneye-project/motioneyeos #261: Obtain Key Google Drive

wolverine-forever, 2016-04-23:
I tried to obtain a new key after i wanted to split the 2 cams over 2 accounts.
However now I get this message : {"error": "internal server error"}
when i try to obtain a new key.

Any clue?

Regards, wolverine

ccrisan, 2016-04-23:
Please attach your logs (especially motioneye.log).

wolverine-forever, 2016-04-27:
This is what it gives:

2016-04-27 16:13:27: [motioneye]    ERROR: 'str' object is not callable
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1401, in _stack_context_handle_exception
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1603, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 218, in get
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 158, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 783, in authorize
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 562, in get
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 596, in _load
TypeError: 'str' object is not callable
2016-04-27 16:13:27: [motioneye]    ERROR: 500 GET /config/1/authorize/?service=gdrive&_username=admin&_signature=db0c20ca639d04c1266216xxxxxxxxxxxxxxxxxxx (xxx.xxx.xxx.xxx) 8.50ms

ccrisan, 2016-05-18:
This should be fixed with https://github.com/ccrisan/motioneye/commit/cecb3571165cd5c04c04adba8a169bc3752e2178.

closed 2016-05-18 by ccrisan