motioneye-project/motioneyeos #263: Wifi issues

brodemack, 2016-04-23:
This happens when i try to set up wifi it just keeps rebooting

[image]

ccrisan, 2016-04-23:
Make sure your network can resolve DNS queries. (that is, make sure the server 10.0.1.1 can act as a  DNS server)

brodemack, 2016-04-23:
It can and is Apple Airport Extreme

brodemack, 2016-04-26:
Sorry for the short answer that was my special needs daughter. Is there a
way for me to manually adjust the settings in terminal? And please keep in
mind that I am a novice when it comes to this. My only real experience has
been with rasbian.

ccrisan, 2016-04-26:
The only way to get motionEyeOS up and running, even for configuration purposes, is to make sure it has internet access. Now given that your wifi connection is allegedly misconfigured on your motionEyeOS system, you'll need to use a wired connection first. Then, accessing the UI should help you enter the network SSID and PSK.

brodemack, 2016-04-26:
Everything works great while it's on the wired network. The problem starts
after we enter the wifi settings in the UI that's when it just keeps
rebooting.

brodemack, 2016-04-30:
Just a note to let you know we figured out what was wrong. We changes the
routers ssid from Bryan's to Bryan and it works great seems that the ' was
the problem all along. We love your product and thank you for it!

Bryan

ccrisan, 2016-04-30:
I'm glad you found the problem. I'll try to see if the apostrophe case can be worked around.

closed 2016-04-30 by ccrisan