motioneye-project/motioneyeos #265: where do I find the meye-xxxxxxxx boot login?

stavropn, 2016-04-25:
I have tried to boot but always stuck to boot login. Where can I find username and password for boot login?
Than you

ccrisan, 2016-04-25:
See https://github.com/ccrisan/motioneyeos/wiki/FAQ#command-line--shell-access,  https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-am-only-getting-white-on-black-text-on-my-monitor-where-is-my-ui and https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-are-the-default-web-ui-credentials.

closed 2016-04-25 by ccrisan

stavropn, 2016-04-25:
Thank you for your prompt answer but I still do not understtand. The program does not boot for all the above to happen.

al090187, 2016-04-25:
How do you know the program has not started?

stavropn, 2016-04-25:
it keeps asking for login

al090187, 2016-04-25:
On what? Are you trying to SSH in, or on the webpage?

al090187, 2016-04-25:
In either case, the default username is admin, with a blank password, as given in the third link ccrisan posted about an hour ago.

stavropn, 2016-04-25:
it keeps saying wellcome   meye-18f00912 login:

al090187, 2016-04-25:
Have you tried admin?

stavropn, 2016-04-25:
yes 
thank you