motioneye-project/motioneyeos #266: Display broken with release 20160410

rv32, 2016-04-25:
Hi,

After tuning a few settings in the new release, the web page display became strange (as without css, see attached image) and without any possibility of doing an action
[image]
If I remember, the last modified setting before crash was changing time from HTTP to NTP.

ccrisan, 2016-04-26:
Could have the card gotten corrupted? You'll need to somehow recover the logs from the SD card and see what had been going on.

hxpaul, 2016-04-26:
maybe a dumb answer @rv32 but have you tried refreshing? I've seen this, the html of the page is in the browser cache, the css is not (somehow) and the motioneyesos server was in the process of rebooting, because you changed a setting. It came back for me quickly when it had rebooted.

triforce, 2016-04-26:
Use CTRL+F5 to force a browser cache refresh.

rv32, 2016-04-26:
Hi,

SSH & FTP are still working.
I tried to reboot the raspi2 : no effect
I tried to update web browser, & even used another one : the same.
The only way to fix this is to format SD card & install again motioneyeos.
... but the problem comes back when I apply the settings.
Maybe there are some interesting log file to understand this issue?
Where?

ccrisan, 2016-04-26:
Please copy all your logs from /var/log/ (via SFTP if you say it still works) and attach them here. It would be interesting to see what changes you make to the settings when you get this error.

rv32, 2016-04-26:
Here it is :
[log motioneyeos 2016 04 25.zip](https://github.com/ccrisan/motioneyeos/files/237315/log.motioneyeos.2016.04.25.zip)

ccrisan, 2016-04-26:
boot.log says:

     # Interface eth0 has IP address 192.168.0.133/24
     # Default gateway is 192.168.0.1
     # DNS server address is 192.168.1.1 192.168.0.1

motioneye.log says:

    2016-04-19 23:01:49: [motioneye]  WARNING: 404 GET /192.168.0.1/static/js/main.js (192.168.0.111) 7.17ms

Somehow motionEyeOS gets the impression it has 192.168.0.1 yet it doesn't. Please attach your /data/etc/motioneye.conf file as well.

rv32, 2016-04-26:
Here it is :
[motioneye.zip](https://github.com/ccrisan/motioneyeos/files/237468/motioneye.zip)
Raspi is not directly connected to 192.168.0.1 but through a router in wifi repeater mode IP 192.168.0.3

ccrisan, 2016-04-26:
Are you sure you're directly accessing your motionEyeOS using 192.168.0.133, regardless of your complex network configuration?

rv32, 2016-04-26:
Yes I'm sure. 192.168.0.133.
Not so complex ! 1 main router with repeaters due to wifi low coverage.
The raspi is wired to the repeater which communicate in wifi to the main router. Other devices are connected to this repeater without any problem.
It worked perfectly before ... maybe I will test with a very new SD card

rv32, 2016-05-21:
Hi,

I've installed a new SD card : everything perfect !!!

Thanks.

ccrisan, 2016-05-22:
Great, I'll close the issue then.

closed 2016-05-22 by ccrisan