motioneye-project/motioneyeos #267: Will the new CSI RPI camera v2 module work?

ReonBalisty, 2016-04-25:
Will the new CSI RPI camera module v2 work with RPI 3 with the Motion Eye OS?

ReonBalisty, 2016-04-25:
Specifically the NoIR camera module

ccrisan, 2016-04-26:
No idea :) I'd have to get one myself but I suppose the worst case scenario is a RPi firmware update requirement which I'll be happy to do.

chumleybob, 2016-04-26:
just tried the new v2.1 camera and the image does not look as good as the old 5mp one 

also its upside down and mirror by default

ReonBalisty, 2016-04-26:
Ordered both will post image differences once I get the old and new set up.

deruiter, 2016-04-29:
I'm using a RPi 3 and the V2 camera. 

Very sad to see the image mirrored nd no option to correct it.

cjmair, 2016-04-30:
I found that using the command :-
v4l2-ctl --set-ctrl horizontal_flip=1
via a SSH terminal connection, changed the image to be the correct way

regards
Colin

cjmair, 2016-04-30:
I also added the command
v4l2-ctl --set-ctrl horizontal_flip=1
to the file /etc/rc.local just before the exit 0 line, to ensure that the command was run at every reboot
Sure there must be a better way, but it works :-)

pieterpoorthuis, 2016-05-10:
I have a Raspberry Pi model 3B with the v2 camera. Problem is that motioneye doesn't recognize the camera. I'm sure the v2 camera works, because other tools like Rpi cam web interface do work with this camera.
I ran rpi-update, so my system is up to date. Motion eye shows the following screen:
[image]

Any idea how to get my v2 camera working in motioneye?

ccrisan, 2016-05-10:
If you ran `rpi-update` and it succeeded, you are probably running Raspbian with motionEye, not motionEyeOS. If you run Raspbian, make sure you have loaded the `bcm2835-v4l2` module.

pieterpoorthuis, 2016-05-10:
thanks, the v2 camera now works. Setted resolution at 1600x1200, motion uses 40% cpu... quite steep

closed 2016-05-10 by ccrisan

olavt, 2016-05-30:
What was the trick to get the v2 camera to work with the MotionEyeOS? I get the "(no cameras)" message.

pieterpoorthuis, 2016-05-30:
Run rpi-update, followed by modprobe bcm2835-v4l2

olavt, 2016-05-30:
Got "command not found" on "rpi-update"

ccrisan, 2016-05-31:
@olavt wait for the next motionEyeOS update. I'm working on it :)

Renader, 2016-06-25:
Hello @ccrisan, thanks for your great work. Is the v2 camera now running with MotionEyeOS? 
I'm new to this and not familiar with motionEye. If the v2 camera doesnt work yet with MotionEyeOS, can i simply use https://github.com/ccrisan/motioneye on rasbian to get the same surveillance features?

ccrisan, 2016-06-25:
It does.

nextgen1979, 2016-07-05:
Hello , i have the same issue with motionEyeOS  , i have the motioneyeos-raspberrypi3-20160703 version and one v2 camera.   and Local Camera  - no camera .  what the solution ?

Thanks in advance

ccrisan, 2016-07-05:
@nextgen1979 does it work with Raspbian?

nextgen1979, 2016-07-05:
i dont have try for the moment , have try only with motioneye OS. I will try tomorrow with raspbian and motioneye application.

ccrisan, 2016-07-05:
Please try raspbian and the raspistill/raspivid applications, just to be able to rule out a faulty camera or any other hardware-related issue.

nextgen1979, 2016-07-05:
ok i will try this tomorrow and keep you inform .  thnks for this great app and os  :)

nextgen1979, 2016-07-07:
Hello, i have make a mistake with the cable of the camera . Sorry .  And thanks for your help and support.

ian6301, 2017-10-10:
I have a clean install of the latest version of MotionEyeOS on my RPi3 and it won't recognise my v2 Pi Camera...

Did this definitely get fixed in the end?

I have checked the camera with Raspbian and it works fine, so not a hardware issue...?