motioneye-project/motioneyeos #268: how to send log file

guadacorte, 2016-04-26:
Hi. I have to screen shots and a motion.log to send in reference to my problem with the ip cam indeanext. Where should I send them. Thanks

ccrisan, 2016-04-26:
You should simply attach them here when posting a comment.

guadacorte, 2016-04-27:
http://i.imgur.com/BdYWYqM.jpg
http://i.imgur.com/PIBegGi.jpg
log
using motion binary /usr/bin/motion-mrdave
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.0.145:554/live/ch0/)
[1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://admin:na602978ac@192.168.0.145:554/live/ch0/): Server returned 404 Not Found
[1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

guadacorte, 2016-04-28:
Any ideas ccrisan? Thanks

ccrisan, 2016-04-28:
Nope. I don't really use rtsp cameras. Try to seek help with the motion guys.

Banditen01, 2016-04-28:
iSpy says rtsp://192.168.0.145/live/ch0 for indeanext cam, could be worth a try without port no !? https://www.ispyconnect.com/man.aspx?n=ideanext#

Banditen01, 2016-04-28:
Have you tried to stream rtsp://admin:xxxxx@192.168.0.145:554/live/ch0/ in for ex. VLC player if it works in VLC it's correct rtsp adress

guadacorte, 2016-04-28:
Hi Banditen01. yes the address works in vlc. I test that yesterday. I have tried taking out the port number but now the log looks like this

1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.0.145/live/ch0/)
[1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://admin:xxxxx@192.168.0.145:554/live/ch0/)

so in this case motioeye os is forcing in port 554

I have tried rtsp://admin:xxxxx@192.168.0.145:554/live/ch0/ but motioneye os still needs something in the username and password fields

Thanks

guadacorte, 2016-04-28:
ccrisan I am a bit confused. I thought that you wrote this software. Why do i need to ask the motion guys? Thanks

Banditen01, 2016-04-28:
Your second picture shows that you have connection with cam, otherwise last window should not be auto-filled in. Is there a lower resolution rtsp adress for ex. ch1 at the end!?

[URL]http://i.imgur.com/PIBegGi.jpg[/URL]

guadacorte, 2016-04-28:
tried ch1,2,3,4 no luck. Also tried other combinations.
The thing is that motioneye os does connect to the ip camera

ccrisan, 2016-04-28:
> ccrisan I am a bit confused. I thought that you wrote this software. Why do i need to ask the motion guys? Thanks

@guadacorte what do you mean by "this software"? I couldn't possibly write every piece of software used by motionEyeOS :)

To quote the [wiki page](https://github.com/ccrisan/motioneye): *motionEye is a web-based frontend for motion.* The motion daemon is definitely not my work!

Banditen01, 2016-04-28:
Have you tried:

/onvif1
/11
/live/ch0_0.h264

Instead of /live/ch0 could be worth a try maybe !?

guadacorte, 2016-04-29:
Sorry ccrisan I should have said frontend software.

Banditen01, 2016-04-29:
> Sorry ccrisan I should have said frontend software.

Quadacorte, 

MotionEyeOS is only a cosmetic frontend software who dresses motion with a nice package without a Linux regular installation, the engine for your CCTV system is Motion and nothing Ccrisan can support

Ask the motion guys:

http://www.lavrsen.dk/foswiki/bin/view/Motion/WebHome

guadacorte, 2016-04-29:
Thanks Banditen01. think i might abandon the idea of using an ip cam and go back to just using usb web cams. Thanks for your help

closed 2017-06-01 by bortek