motioneye-project/motioneyeos #269: LD_PRELOAD=/usr/lib/v4l2convert.so

ivityc, 2016-04-27:
Hi 
I used motion previously with RPI3 and a Creative USB Webcam. To have it working I had to start the process with the  "LD_PRELOAD=/usr/lib/v4l2convert.so" before it.

Is there a way to configure motioneye to do the same ?
I tried to modify the /usr/bin/meyectl file and motion file with no luck

Thanks for your support
Iv

ivityc, 2016-04-28:
Ok I managed to edit the /etc/init.d/S85motioneye
adding the following lines and now the webcam works fine

  LD_PRELOAD=/usr/lib/v4l1compat.so
  export LD_PRELOAD

closed 2016-04-28 by ivityc

djtale, 2017-12-28:
Hello @ivityc , do you remember where in S85motioneye you put those lines ?

That would help me a lot.

ivityc, 2017-12-28:
hi, I do not remember exactly, but I think it was quite at the beginning of the file right after the line
#!/bin/bash