motioneye-project/motioneyeos #270: RPi3 wifi not working

bobbydriver, 2016-04-27:
New RPi3 using motioneyeos-raspberrypi3-20160410.img build

Boot with wired connection. Works fine and set the wifi SSID and PSK, then reboot without wired connection and it fails to establish the wifi connection

Also the SSH keys fail to load at boot - so even when it is wired I cannot SSH to check config.

SuneBielefeldt, 2016-04-27:
# **Raspberry Pi 3 with Motion Eye Os**
I have a few weeks ago installed an RPI 3 with Motion Eye Os. I used Image motioneyeos-raspberrypi3-20160320.img.gz.
There were no problems, and I could effortlessly set up Wi-Fi, and disconnect the Ethernet cable.

Subsequently (a few weeks later) I have upgraded to 20,160,410 release, and it still functions fine. No restart or other problems.

I therefore encourage you to try 20160320 release, see if it works, and then upgrade to 20160410 if everything is okay.

Good luck with this amazing system, I think you will be happy about it, and Calin Crisan's great effort to develop it. :-)

[image]

ccrisan, 2016-04-27:
@bobbydriver the SSH keys problem usually happens when the device power is cut soon after the very first boot. The filesystem doesn't get synched to the SD card and hence the missing or corrupted files. In these cases I recommend rewriting the OS from scratch. I know a fix for preventing these issues but I haven't managed to implement it yet.

ccrisan, 2016-04-27:
@SuneBielefeldt thanks for the nice screenshot!

bobbydriver, 2016-04-27:
Hi - just tried using the previous image motioneyeos-raspberrypi3-20160320.img.gz but the same thing happens. Is fine until I switch to the wireless connection. I can't even work out what is going on with the network config as I can't find /etc/network/interfaces 

Am I doing the right steps? Burn SD card with img file. Connect wired - all works OK, camera is showing images. Then switch the Wifi on in the web page, enter the SSID and Passcode, Restart device - still wired at this point and it comes back up on the wired IP address. So I can check that it has saved the wireless config in the GUI. Then shutdown via the GUI. Unplug the Ethernet and restart the Rpi3. It does not connect, and even reconnecting the Ethernet again for a second reboot does not bring back the wired connection either?!

bobbydriver, 2016-04-28:
Ah OK - found the description of how the networking gets set up at https://github.com/ccrisan/motioneyeos/wiki/Tweaks - will have a look at that later and get back to you

bobbydriver, 2016-04-28:
So the wlan0 seems to be setting up OK, but it's not finding any DHCP lease offers after trying 5 times, so gives up

> Jan  1 00:10:04 meye-9eadbcf6 daemon.info dhclient: No DHCPOFFERS received.

> Jan  1 00:10:04 meye-9eadbcf6 daemon.info dhclient: No working leases in persistent database - sleeping.

Also in dmesg log I spotted this - not sure if it matters

> [Thu Jan  1 00:00:01 1970] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists

SuneBielefeldt, 2016-04-28:
**# Network settings:**
Try to set up static IP address.
It must lie outside the range as your router assigns IP addresses.

Eg. 192.168.1.15 will be used, I believe. Frequently used in a range of from 30 or 100 and 100 addresses onwards. It is called DHCP scoop

It is important that the first 3 segments are the same throughout the network.
Mask: 255.255.255.0
Gateway: Your Routers IP adress.

From a Windows PC's Command prompt type **c: \> ipconfig / all **and you will get all information.

Example:

Ethernet adapter Ethernet:

   Connection-specific DNS Suffix  . : mydomain
   Description . . . . . . . . . . . : Intel(R) 82567LM-3 Gigabit Network Connection
   Physical Address. . . . . . . . . : 00-0F-FE-C5-FC-BA
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::7c18:8f6d:924b:d30b%7(Preferred)
   IPv4 Address. . . . . . . . . . . : 192.168.1.195(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Lease Obtained. . . . . . . . . . : 28. april 2016 torsdag 12:24:33
   Lease Expires . . . . . . . . . . : 29. april 2016 fredag 06:24:28
   **Default Gateway . . . . . . . . . : 192.168.1.1**
   DHCP Server . . . . . . . . . . . : 192.168.1.1
   DHCPv6 IAID . . . . . . . . . . . : 50335742
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-1E-3D-97-93-00-0F-FE-C5-FC-BA
   DNS Servers . . . . . . . . . . . : 8.8.8.8

Default Gateway is your Router's IP adress.

By using a static / fixed IP address, you always know which IP address your Motioneye have when you need to connect.

I always use this method on my Motioneye Cam's, and it works.

Good luck.
Sune Bielefeldt

SuneBielefeldt, 2016-04-28:
The first time I had set up MotionEyeOs, I also had a lot of questions, and could not get it in the air. But when it finally succeeded, it has just been running for over half a year.

The first step is always the hardest, but you'll get it to succeed.

bobbydriver, 2016-04-28:
Thanks - will carry on digging. It's weird as I had it running fine on an old Raspberry Pi A with wifi dongle - straight from the image,  but the board was so under-powered that it really struggled. Now got a Pi 3 and hoping to take advantage of the better spec. Will try static IP

bobbydriver, 2016-04-30:
Still not working. Tried to change to a Static IP for wifi , but that doesn't work either.

Then if I try to reconnect via wired connection after this, the device shows up in my router as connected on a wired DHCP IP address, but doesn't load the web interface or allow SSH so again the only option is to reflash the card back to the start. Doesn't ping either

Running out of ideas now. Did you say there was an issue with special characters in the passkey? I have an @ symbol in my password. Could that be an issue?

SuneBielefeldt, 2016-04-30:
**# Wi-Fi Networking dont work yet**

After setting up Wi-Fi network, you may keep ethernet cable connected after reboot.

So you can connect both Ethernet IP address and Wi-Fi address from your browser.

Thus you can also correct any. errors or try other options without having to reimage.

Try to set up a simple user name and password for your Wi-Fi router so you are sure it is not special characters that are the problem.

It could easily be @ as used in the password that is your problem.

You have fought bravely, often it is the simple things which is why it does not work.
Keep hope and faith, things can work.

Good luck :-)

bobbydriver, 2016-05-01:
Ok, so I still don't have the wifi working on my Raspberry Pi 3. 

I managed to get the static IP set on both wired and wireless connection, but it just doesn't seem to connect to the wifi. 

In which startup script does it read the SSID and PSK from wpa_supplicant.conf and try to connect to the wifi? How can I test this from the command line. I'm not seeing anything in the logs to indicate a problem. Really need someway to debug this as it looks to be a general problem with the Pi3 build

bobbydriver, 2016-05-02:
OK - have found the root cause.

Did 

> iwlist wlan0 channel

and it is only seeing my wifi channels 1-11. Guess what? My wifi is running on channel 12...hence it's not being picked up by your motioneyeos build. It's a known problem with old RPI3 kernels

https://github.com/raspberrypi/linux/issues/1355

Obviously I don't want to change this as I've found this to be the most stable channel in my neighbourhood. So I looked into the solutions and for a standard RPI3 it seems simple enough, but none of these commands to upgrade the RPI3 kernel work in your motioneyeos build.

I don't know much at all about buildroot, but I guess you would have to create a new rpi3 img with the latest kernel in it for me to be able to use motioneyeos over wifi

I hope this is in your plans to fix and not too big a deal?

ccrisan, 2016-07-03:
The moment this gets fixed in Raspbian (if it hasn't been done already) it will be fixed in motionEyeOS as well :)

closed 2016-07-03 by ccrisan

luckydog1, 2016-08-08:
Hello Ccrisan, Can you please tell me if the fix you mentioned on the 3rd of July has been implemented as I am having the same problem with Motioneye not connectinng to WIFI,,at 79 years I find it difficult understand everything,I had no difficulty with  motionPi,I would be grateful for your assistance, regards aussiejohn

ccrisan, 2016-08-08:
I'm not sure if it has been already fixed in Raspbian and I have no means to test it. All I can say is that the next motionEyeOS version will surely use the latest kernel (and drivers) available for Raspbian at that time. And no, I have no ETA for the new version yet.

luckydog1, 2016-08-08:
Thanks very much for your prompt reply,I will look out for the new version.

Best Regards

aussiejohn

FingolSE, 2016-09-12:
Ahhh finally in this thread I found the root cause for my problems with connecting with one of my WiFi nodes - channel 13. During the search I changed to channel 12 of course without success. But at the same time I have been running the latest raspian and have no problems in connecting to the troublesome network with channel 13. Did I understand it correctly that motioneye is built upon raspian ? which version ?

Same this for me the air on the 2.4 GHz is extremly crowded and channel 12 or 13 is my only option. Anyone found a way around this problem ? (without getting a 5GHz wifi stick ;) )

/Fingol

ccrisan, 2016-09-12:
Well the kernel commit id can always be found in [this](https://github.com/ccrisan/motioneyeos/blob/master/configs/raspberrypi3_defconfig) file:

    BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="https://github.com/raspberrypi/linux/archive/bc6fd81001d530f8785245735cb11081d3cc06f2.tar.gz"

FingolSE, 2016-09-12:
Findings: The region domain of the wifi is unset and therefore only the global channel set is allowed (1-11).

Solution: If the region domain is set to as in my case Sweden: SE then voila it works and all channels 1-13 is usable. Achieved by adding the following row "options cfg80211 ieee80211_regdom=SE" in the file: /etc/modprobe.d/modprobe.conf

ccrisan, 2016-09-12:
Thanks for finding the root cause of this problem. Could you please add the line `cfg80211 ieee80211_regdom=SE` (no `options`!) to a file called `/data/etc/modules` and see if it works as expected?

You could also use `/data/etc/modprobe.conf` and add your entire line there (with `options`).

The files that are not on `/data` will be overwritten by a system update.

FingolSE, 2016-09-17:
Tested your alternative way of adding a file "/data/etc/modules" with the following line as the only information: "cfg80211 ieee80211_regdom=SE". Works perfect, thanks!

ccrisan, 2016-09-17:
Thanks for getting back to me. Do you have any idea what would be the best approach of fixing this generally? How does Raspbian handle this issue?

FingolSE, 2016-09-17:
Raspian have in the raspi-config utility a setting under the Internationalisation option where you set the WiFi country. It's a looong list, unfortuatly the regulations are country specific so for instance EU is not a valid option. I see two options...
1. In your Network options also add this long list of country setting.
2. Default choose a country (UK will work) that has atleast ch 1-13 open. If MotionEye is used in a country where the channel 12-13 is a no go (read US) the system will not output any significant radio energy on those frequencies, you only use this setting to be able to search for the user selected access point.

ccrisan, 2016-09-17:
Alright then I'll go with the second one :) Thanks for the info.

FingolSE, 2016-09-17:
When I double checked it seems there are information of either UK or GB as valid country code, perhaps both will work but not sure.

ccrisan, 2016-10-26:
63014656a10706c3e268d3b6d03086018471d7fb adds support for specifying the country for the wifi regulatory domain. The variable [os_country](https://github.com/ccrisan/motioneyeos/wiki/os.conf#os_country) from the /data/etc/os.conf file should be edited. It now defaults to `US`.

riahc3, 2016-11-27:
Downloaded the latest version and once I switch to wireless, I cannot access it anymore. Not by wifi, ethernet or anything. I have to rewrite the image.

ccrisan, 2016-11-27:
Are you trying to access it using the same IP address you got when using the wired connection?

riahc3, 2016-11-27:
Yes. I am also trying to search for it via nmap and nothing.

ccrisan, 2016-11-27:
Try using the Fing android app. It should detect the *new* IP address your PI got on its wifi interface.

riahc3, 2016-11-27:
Nothing. Cannot see it using fing and/or nmap

ccrisan, 2016-11-27:
Can you connect a monitor to it so at least we can know for sure the wifi does not connect at all?

riahc3, 2016-11-27:
Rewriting the image and booting....

riahc3, 2016-11-27:
Rewrote the image and no boot BUT it does connect to the IP thru ethernet...

ccrisan, 2016-11-27:
More details? Did you connect a monitor? What do you mean by "no boot" and "connect to the IP"?

riahc3, 2016-11-27:
Yeah, sorry.

Fresh image, I connected the monitor and NOTHING shows up. Doesnt matter if its ethernet or not. No video shows up. 

I cannot connect to my ETHERNET IP.....

ccrisan, 2016-11-27:
If nothing shows up, that's a bit odd. Then your monitor must not be connected correctly or compatible at all.

riahc3, 2016-11-27:
Raspbian works.

ccrisan, 2016-11-27:
Wait what? If you cannot connect using ethernet, then the OS image hasn't been correctly written. See https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-wont-boot.

riahc3, 2016-11-27:
Yes, I CAN connect with ethernet. Writing a FRESH image, WITH ethernet, I see NO video.  With Raspbian I have no issues.

riahc3, 2016-11-27:
Do you think changing "os_country" might help? I do not live in the US.

ccrisan, 2016-11-27:
Changing the country shouln't solve the issue. Have you tried other versions of motionEyeOS?

riahc3, 2016-11-27:
No. That was my next idea....Trying out the previous version.

Im on a PI 3.