motioneye-project/motioneyeos #277: How to install

dmarkscouk, 2016-04-29:
I'm on OSX and trying to use the shell script. I've tried the following but just keep getting errors with the SD card mounded and unmounted.

I'm running this from the director with the shell script and img in it.

sudo ./writeimage.sh -d /dev/disk2 -i “motioneyeos-raspberrypi3-20160410.img”
sudo writeimage.sh -d /dev/disk2 -i “motioneyeos-raspberrypi3-20160410.img”
sudo ./writeimage.sh -d /dev/disk2 -i motioneyeos-raspberrypi3-20160410.img
sudo writeimage.sh -d /dev/disk2 -i motioneyeos-raspberrypi3-20160410.img
sudo bash ./writeimage.sh -d /dev/disk2 -i “motioneyeos-raspberrypi3-20160410.img”
sudo bash writeimage.sh -d /dev/disk2 -i “motioneyeos-raspberrypi3-20160410.img”
sudo bash ./writeimage.sh -d /dev/disk2 -i motioneyeos-raspberrypi3-20160410.img
sudo bash writeimage.sh -d /dev/disk2 -i motioneyeos-raspberrypi3-20160410.img

Any ideas much appreciated.

Thanks,

David.

ccrisan, 2016-04-29:
There is a known problem with OSX and the mount command. I'll look into it. Until then, I suggest using other more common methods of getting the image to your SD card.

ccrisan, 2016-04-29:
Actually strike that. @pauly was nice enough to fix this already: https://github.com/ccrisan/motioneyeos/pull/258. Just download the latest [writeimage.sh](https://github.com/ccrisan/motioneyeos/blob/master/writeimage.sh) and you should (hopefully) be fine.

dmarkscouk, 2016-04-29:
Hi @ccrisan thanks for your replies. I've been using the latest version from what I can see. It says the last time the writeimage.sh was updated was 9 days ago. I downloaded this today, which should mean I'm using the latest version.

dmarkscouk, 2016-04-29:
I'm getting the following...

`Last login: Fri Apr 29 21:15:24 on ttys000
Davids-MacBook-Air-2:~ dmarks$ cd eye
Davids-MacBook-Air-2:eye dmarks$ sudo bash ./writeimage.sh -d /dev/disk2 -i motioneyeos-raspberrypi3-20160410.img
:: writing disk image to sdcard
dd: /dev/disk2: Resource busy
:: mounting sdcard
mount_msdos: /dev/disk2s1: Resource busy
:: unmounting sdcard
umount: ./.boot: not currently mounted
rmdir: ./.boot: Directory not empty
:: you can now remove the sdcard
Davids-MacBook-Air-2:eye dmarks$ `

ccrisan, 2016-04-30:
@dmarkscouk I merged the fix just minutes before answering to your

dmarkscouk, 2016-04-30:
Hi @ccrisan . Thanks again, but I'm getting the same error. I've re-downloaded the file (and tried pushing shift refresh before doing so to make sure it's not a cached version on the site, but am still getting the same error. When I look at write.sh online it say's it's 10 days old, so maybe for some reason, I'm not getting the latest version?

[image]

[image]

Thanks again, David.

dmarkscouk, 2016-04-30:
In aid of working out if I have the correct latest writeimage.sh file, This is the contents of the writeimage.sh file...

#!/bin/bash -e

function usage() {
    echo "Usage: $0 [options...]" 1>&2
    echo ""
    echo "Available options:"
    echo "    <-i image_file> - indicates the path to the image file (e.g. -i /home/user/Download/file.img.gz)"
    echo "    <-d sdcard_dev> - indicates the path to the sdcard block device (e.g. -d /dev/mmcblk0)"
    echo "    [-n ssid:psk] - sets the wireless network name and key (e.g. -n mynet:mykey1234)"
    echo "    [-s ip/cidr:gw:dns] - sets a static IP configuration instead of DHCP (e.g. -s 192.168.1.101/24:192.168.1.1:8.8.8.8)"
    exit 1
}

if [ -z "$1" ]; then
    usage
fi

if [[ $(id -u) -ne 0 ]]; then echo "please run as root"; exit 1; fi

function msg() {
    echo ":: $1"
}

while getopts "a:d:f:h:i:ln:o:p:s:w" o; do
    case "$o" in
        d)
            SDCARD_DEV=$OPTARG
            ;;
        i)
            DISK_IMG=$OPTARG
            ;;
        n)
            IFS=":" NETWORK=($OPTARG)
            SSID=${NETWORK[0]}
            PSK=${NETWORK[1]}
            ;;
        s)
            IFS=":" S_IP=($OPTARG)
            IP=${S_IP[0]}
            GW=${S_IP[1]}
            DNS=${S_IP[2]}
            ;;
        *)
            usage
            ;;
    esac
done

if [ -z "$SDCARD_DEV" ] || [ -z "$DISK_IMG" ]; then
    usage
fi

function cleanup {
    set +e

    # unmount sdcard
    umount ${SDCARD_DEV}* >/dev/null 2>&1
}

trap cleanup EXIT

BOOT=$(dirname $0)/.boot

if ! [ -f $DISK_IMG ]; then
    echo "could not find image file $DISK_IMG"
    exit 1
fi

gunzip=$(which unpigz || which gunzip)

if [[ $DISK_IMG == *.gz ]]; then
    msg "decompressing the gzipped image"
    $gunzip -c $DISK_IMG > ${DISK_IMG::-3}
    DISK_IMG=${DISK_IMG::-3}
fi

umount ${SDCARD_DEV}* 2>/dev/null || true
msg "writing disk image to sdcard"
dd if=$DISK_IMG of=$SDCARD_DEV bs=1048576
sync

if which partprobe > /dev/null 2>&1; then
    msg "re-reading sdcard partition table"
    partprobe ${SDCARD_DEV}
fi

msg "mounting sdcard"
mkdir -p $BOOT

if [ `uname` == "Darwin" ]; then
    BOOT_DEV=${SDCARD_DEV}s1 # e.g. /dev/disk4s1
    umount ${SDCARD_DEV}* 2>/dev/null || true
    mount_msdos $BOOT_DEV $BOOT
else # assuming Linux
    BOOT_DEV=${SDCARD_DEV}p1 # e.g. /dev/mmcblk0p1
    if ! [ -e ${SDCARD_DEV}p1 ]; then
        BOOT_DEV=${SDCARD_DEV}1 # e.g. /dev/sdc1
    fi
    mount $BOOT_DEV $BOOT
fi

# wifi
if [ -n "$SSID" ]; then
    msg "creating wireless configuration"
    conf=$BOOT/wpa_supplicant.conf
    echo "update_config=1" > $conf
    echo "ctrl_interface=/var/run/wpa_supplicant" >> $conf
    echo "network={" >> $conf
    echo "    scan_ssid=1" >> $conf
    echo "    ssid=\"$SSID\"" >> $conf
    if [ -n "$PSK" ]; then
        echo "    psk=\"$PSK\"" >> $conf
    fi
    echo -e "}\n" >> $conf
fi

# static ip
if [ -n "$IP" ] && [ -n "$GW" ] && [ -n "$DNS" ]; then
    msg "setting static IP configuration"
    conf=$BOOT/static_ip.conf
    echo "static_ip=\"$IP\"" > $conf
    echo "static_gw=\"$GW\"" >> $conf
    echo "static_dns=\"$DNS\"" >> $conf
fi

msg "unmounting sdcard"
sync
umount $BOOT
rmdir $BOOT

msg "you can now remove the sdcard"

ccrisan, 2016-04-30:
@dmarkscouk, unfortunately that is the fixed version. Maybe @pauly can help us find a new fix for this.

hxpaul, 2016-05-02:
you have a similar issue to me but at a different point in the code, will see if I can recreate it.

dmarkscouk, 2016-05-02:
Brilliant thankyou.

dmarkscouk, 2016-05-05:
@hxpaul any luck with recreating the issue?
Thanks, David.

hxpaul, 2016-05-09:
sorry not had time to try and recreate this yet. You could try taking a local copy of https://raw.githubusercontent.com/ccrisan/motioneyeos/master/writeimage.sh and tweaking it, drop some debugging in. I'll be getting another pi and setting up with motioneyeos "soon".

dmarkscouk, 2016-05-09:
Thanks Paul. I'm very new to any of this Raspberry Pi etc. I'm not sure
where I would start with tweaking it / or how to add debugging in?

dmarkscouk, 2016-05-10:
I've managed to install this pretty easily with a bit more tinkering. In case it's useful to anyone else here's how...

cd /eye
- This is the directory in which I downloaded the .img file.

diskutil unmount /dev/disk2s1
- Unmount the partition on the SD card. You'll need to adjust this to make sure you're targeting the SD card and not another drive.

sudo dd if=motioneyeos-raspberrypi3-20160410.img of=/dev/rdisk2 bs=1m
- The command to copy the .img file to the SD card. As above you need to adjust the disk name. Once you typed this you will need to enter the password for OSX.

hxpaul, 2016-05-19:
I just tried again, got a raspberry pi zero and going to try that out. Completely forgot how to do this. In the end:
```
$ sudo umount /dev/disk3s1
umount: /dev/disk3s1: not currently mounted
$ diskutil unmountDisk /dev/disk3
Unmount of all volumes on disk3 was successful
$ sudo ./writeimage.sh -d /dev/disk3 -i motioneyeos-raspberrypi-20160410.img -s "192.168.1.104/24:192.168.1.254:8.8.8.8" -n "myssid:mykey"
```

so fingers crossed

wormix, 2016-05-19:
I had the exact same problem trying to install a Zero using Mountain Lion. Switched to my Snow Leopard Version and got further, but then had issues remounting card at the end.  Switching Sd card adapter fixed that. Unfortunately no onboard camera support for the zero yet, but should be ok on USB.

ashrimplin, 2016-06-14:
Having same issue - resolved using [RPi-sd card builder,](https://alltheware.wordpress.com/2012/12/11/easiest-way-sd-card-setup/) but unable to set up static IP. Does anyone know if there are commands to run where I can set static IP post installation?

ccrisan, 2016-06-14:
You could access the UI once to configure a static IP.

ashrimplin, 2016-06-14:
D'oh :< Sorry man. I overlooked it. Thank you!

shaungrady, 2016-07-07:
I'm getting this error:

```
:: decompressing the gzipped image
./writeimage.sh: line 83: -3: substring expression < 0
```

jwenz723, 2017-02-03:
I still had issues with the writeimage.sh script not working on my MAC.  I just submitted a pull request that will hopefully fix these issues for other people running MAC OS X as well.

https://github.com/ccrisan/motioneyeos/pull/748

bortek, 2017-04-11:
I can see that the pull request was merged and consider this working. Will close this Issue for now.

closed 2017-04-11 by bortek

andrewclapham, 2021-02-24:
I had this issue too, when using Mac OS X and the `writeimage.sh` installation method.

**Here's what solved this for me.**

Step 1: Plug in the USB SD card reader with SD Card.
Step 2: Identify the device. Mine was `/dev/disk2`
Step 3: Don't "Eject" the device. Instead, run `diskutil unmountDisk /dev/disk2` and you'll see the output "Unmount of all volumes on disk2 was successful"
Step 4: Then run the usual `./writeimage.sh ...` installation script.

Hope this helps!

starbasessd, 2021-02-24:
@andrewclapham 
(a) You realize this issue is 4 years old.
(b) was closed 4 years ago.
(c) there have been MANY iterations of the writeimage.sh since then?