motioneye-project/motioneyeos #278: RPi 3 and V2 camera module: mirrored image - how to correct that?

deruiter, 2016-04-29:
Is there a possibility to correct the defaut mirrored image of the V2 camera module?
Thanks!

cjmair, 2016-04-30:
I found that using the command :-
v4l2-ctl --set-ctrl horizontal_flip=1
via a SSH terminal connection, changed the image to be the correct way

regards
Colin

cjmair, 2016-04-30:
I also added the command
v4l2-ctl --set-ctrl horizontal_flip=1
to the file /etc/rc.local just before the exit 0 line, to ensure that the command was run at every reboot
Sure there must be a better way, but it works :-)

deruiter, 2016-04-30:
Hello Colin,

Thank you so much for this solution, I searched for hours to find it but I could not.

I use the MotionEyeOS version for my RPi3 which has no rc.local but I will install motionEye manually so I can add the ‘flip’ option.

Sincerely,
Erik de Ruiter

herman56, 2016-05-01:
I tested v4l2-ctl --set-ctrl horizontal_flip=1 but it seems it needs to be corrected to v4l2-ctl --set-ctrl= horizontal_flip=1

deruiter, 2016-05-02:
One question: how can I add the 'v4l2-ctl --set-ctrl horizontal_flip=1' to the MotionEyeOS as this version has no /etc/rc.local file. Sorry for such a dumb question...

cjmair, 2016-05-02:
I would try this.
SSH to the Pi, as root, then use these commands

CD /etc
mount -o remount,rw /
nano rc.local

Use the editor to add the following two lines:

#!/bin/sh
v4l2-ctl --set-ctrl horizontal_flip=1

use Ctrl&O to save the file then Ctrl&X
issue a reboot command

Good Luck

deruiter, 2016-05-02:
Thank you very much for your patience with a newbie...

YES it does work now!
Although I added the line **v4l2-ctl --set-ctrl horizontal_flip=1** to the file **/etc/init.d/S99showinfo**
(I did create rc.local in the /etc directory but nothing happened)

closed 2016-05-02 by deruiter