motioneye-project/motioneyeos #281: EasyCap USB video capture support (add driver)

Birdcam, 2016-05-03:
Hi, I am using MotionEyeOS with the Raspberry Pi 2 & 3 and USB webcams for nestbox camera and I am loving it! It's so easy to setup. Thanks for that! I would like better image quality by using a CCTV camera with USB video capture device but it is not supported.
Is it possible to add support for the EasyCap USB video capture device? I know some versions of Raspbian support it out of the box. I tried to add the drivers myself but the OS is too much secured for that and I am not verry good with Linux.
Thanks!

ccrisan, 2016-05-03:
EasyCAP will be added to the next version.

Birdcam, 2016-05-03:
Thanks! I can't wait.

closed 2016-05-03 by Birdcam

ccrisan, 2016-05-20:
Fixed with 92b32805ebdfdb466b1c2328d5ef4c5ad3ecb571.

Birdcam, 2016-10-17:
I updated to the latest version but my easycap is still not recognised. Do I need to do anything to make it work?

spyhunter99, 2019-03-28:
mine doesn't work either unfortunately. It works fine with vlc, but motion just doesn't capture from it