motioneye-project/motioneyeos #284: Working Schedule not affecting Still Images

scohrab, 2016-05-07:
It appears that the working Schedule only applies to Motion Detection, but it would be really useful for it to apply to Still Images when using 'Interval Snapshot' mode to create timelapse movies. That way, we could easily capture daytime pictures only of, for example, the construction of a building, over a period of weeks or months. Similarly timelapse movies of rush hour traffic, or indeed any number of similar applications.

Alternatively, can anyone suggest a way of configuring MotionEyeOS to do this ?

jbthoma, 2016-10-30:
Yes, please consider this feature!  This would eliminate about 12 hrs of "black screen" pictures my camera takes everyday.  Thanks for the great software!

EelskeM, 2016-12-03:
Same request here: Please consder the possibility to take still images only during working schedule. Great for timelapse pictures  during daylight  (in my case skypictures/wheatherdevelopment during daylight.

bortek, 2017-06-01:
@ccrisan can you please comment on this ?

UrmelCH, 2017-10-18:
Dear all

Are there any news on this topic? It would be a great feature.

Thanks
Urs.

Dudematters, 2020-11-29:
I would like to see this feature, preferable with daylight feature built-in for the location. I've noticed the last post on this is in 2017.

starbasessd, 2021-11-23:
The currently 'easiest' way to handle this is to have 2 settings, one with the capture settings 'on' and one with the capture settings 'off', and use a crontab or action button (or both) to do what Working Hours does, en effect. I've posted similar methods here for other issues.