motioneye-project/motioneyeos #287: question about this screenshot

ts4s, 2016-05-09:
so i ran a pi3 for a week before relocating it to it's future home. upon power up i get this screen, what does it mean? do i need to rebuild the sd card again or is there way to fix it? 

[image]

ccrisan, 2016-05-09:
Corrupt SD card? Make sure the power supply is good enough. At this point, I recommend rewriting the OS image from scratch.

ts4s, 2016-05-09:
so is the motioneye os this unstable? this is the 3rd clean build i've done and I keep ending up at this screen at some point? I use win32diskimager to wire the lastest build to the micro sd card. even went back one version, then perform the upgrade to the lastest version, and still ended up at this point.

al090187, 2016-05-09:
Have you tried another power supply as suggested? It is well known that an insufficient supply of power causes sd card corruption. That is not a problem with the os.

ts4s, 2016-05-09:
I've tried 2 different 5 volt 2.5 amp power supply and even powering it off my laptop. 

If the os is good, then why can I no login with my user name and password? Os shouldn't be loosing that info.

ccrisan, 2016-05-09:
The OS is not that unstable :) My personal opinion is that it's a bit more stable than Raspbian, given the fact that it's mostly read-only. I've been using it for some time now myself on several cameras and I had no stability issues with it.

If you say you've encountered the error you mentioned several times, I recommend trying to replace your components one by one, since it seems to me we're dealing with a hardware-related problem. If you say you had changed your power supply with a 2.5 Amps one, try replacing the SD card (model).

One more thing: try leaving it running as long as possible after the very first boot. Chances are that, if you cut its power immediately after the first boot, the initial settings that are copied over to the data partition are lost.

ts4s, 2016-05-09:
So how long then? Over night? This last build ran a week before I physically cut power. Did do a few reboot via software for the configuration.

ccrisan, 2016-05-09:
10 minutes should be enough.

ts4s, 2016-05-10:
I've have it up and running again, so tonight I will test it again once I reach my destination. 

I'm still confused on why it would boot to a user prompt but not allow the user to login? If it where a corrupt card, then it shouldn't boot to a user login prompt. user credentials only change during setup and is only access during logins and times of updating the credentials.   

Would it be better to have a second storage device connect?

ccrisan, 2016-05-10:
It's hard to predict what part of the system gets corrupted :) Usually it's the data partition as it's the only writable partition. As your admin password is on the data partition, mostikely

ccrisan, 2016-05-10:
However we've been talking about the os itself while I believe the cause is hardware-related.

ts4s, 2016-05-11:
couple of question first before i go into details of testing. 

1. Is the ethernet cable require to be connect all the time or can it be remove once coneected to a wifi connection?

2. is there a better way to check for this connection without rebooting? Just seem problematic after what i been going through.

3. What are the SD card requirements? 

with that being said, the sd card was the issue. When i went to plug everything up, I tried to use the hotel ethernet port (should have known it was dead since everything is wifi nowdays). Watched the pi reboot 3 times before getting it connected to my mac for internet sharing. It cam up and ran a it for a bit before switch to fastcam. prompt for a reboot and blame, got that screen again. grab my 32 GB sandisk micro and loaded it up. restored the config file, this is really handy feature. and was off running. switch back to normal mode without issue. left to grab some dinner and came back with 8gb PNY and ran it through the same steps.

photo of the 2 SD cards
[image] 
PNY Spec
[image] 
emtec spec
[image]  
emtec after failing
[image] 

this is the first os that i've come across this issue with any sd card.

closed 2016-05-11 by ts4s

ccrisan, 2016-05-11:
1. The ethernet cable can be safely removed after a wifi connection has been configured. motionEyeOS requires one Internet connection.

2. I'm not sure what you mean by "checking this connection". You should have an IP address given by your DHCP server, or a static IP, which could be used for administrative purposes as well.

3. There's no list of supported SD cards and I wouldn't know how to put together a set of requirements, as they depend on the board model. If you're on Raspberry PI, you should go looking for what works best with the Raspberry PI. From my personal experience, I never had problems with any of the SD cards I have tried, other than physical damages.