motioneye-project/motioneyeos #288: motionEye not detecting RPI camera

jwlondon98, 2016-05-10:
When i try to add a camera motionEye doesn't detect my RPI camera. The camera is definitely working because i checked it in terminal with the "vcgencmd get_camera" command and it returned "supported = 1 detected = 1". I'm also able to take pictures with raspistill. Anyone have any idea what the problem is?

ccrisan, 2016-05-10:
motionEyeOS or Raspbian + motionEye?

jwlondon98, 2016-05-10:
Well I was having trouble getting motionEyeOS to boot from the burnt img file so I ended up installing it in Raspbian with this tutorial - https://youtu.be/G7DwPzSbLlc

ccrisan, 2016-05-10:
In that case, make sure you have loaded the `bcm2835-v4l2` module.

jwlondon98, 2016-05-10:
Can you please explain how I would go about doing that. Also, do you know the commands to shutdown motion and eye motion. They boot on startup which is fine but in the future I may not want them on since they use significant processing power.

invisan, 2016-05-21:
modprobe bcm2835-v4l2 would be the process to load the module. 
I would suggest thought to use motioneyeos instead of Raspbian. Try to redownload the img and then copy it to your SD via WIN32DiskImager.

closed 2016-07-03 by ccrisan

manthis, 2016-12-20:
I use motioneyeos that I've just installed for the first time and my raspberry pi module camera doesn't seem to be detected. However, bcm2835-v4l2 is loaded while "vcgencmd get_camera" command returns "supported = 1 detected = 1".

ccrisan, 2016-12-20:
How do you know the camera isn't detected?

manthis, 2016-12-20:
@ccrisan I have no image on the web interface. I'm using a RPi Noir Camera. Anything I could try to make it work?

ccrisan, 2016-12-20:
What happens when you try to add a new camera? Screenshots always help.

manthis, 2016-12-20:
When just setup I already have a camera but no image. If I try to add another camera, it says I have no camera. I think the screenshots speak for themselves.

[image]
[image]

ccrisan, 2016-12-21:
In your case the camera is detected by both the OS and motionEye, but it doesn't work, for some reason. Please try using Raspbian and see if it works. It might be broken.

dottired, 2017-01-01:
I was getting the same issue.  Motioneye not detecting my camera even though the camera was functioning (able to take pics n vids via terminal).  

It took a few hours but after an upgrade to motion, it's picking it up fine.

Great work on the interface ccrisan.  Clean, intuitive and so many options.

seanlee77, 2017-01-04:
Hi,

i just installed your build of motion and on a pi2.  I have tested the local camera using raspistill and it take pics and video's with no issues.

I've installed the latest version of Raspbian.  Can you provide any some trouble shooting steps.

Also do you know if a xbox 360 kinect cam will work with this software?

Thanks
S

[image]

armastaja, 2017-01-05:
I've noticed on two different installs that if I click "You have not configured any camera yet. Click here to add one..." I am unable to add a local camera. HOWEVER, if I click on the settings icon in the upper left and the camera dropdown box appears, I am able to click "add camera" and the local camera is recognized.

teddysim, 2017-03-05:
I had the exact same issue as the OP and managed to solve it.
Have a look here: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
On bullet point # 4, at the top of the article labelled "Before Proceeding", the author clearly states that "If you want to use the CSI camera module for the Raspberry PI, you need to add bcm2835-v4l2 to /etc/modules and reboot."

Just `sudo nano /etc/modules` and add the single entry: `bcm2835-v4l2`.

Reboot and you should be able to see the camera module load and be available to select in the drop-down menu.

iamkeq, 2017-03-28:
I attempted to: 
Just sudo nano /etc/modules and add the single entry: bcm2835-v4l2.

But I am met with the error that the sudo command is not found.

I am using Putty to log into the Pi and using the admin username with no password which allowed me to get into the file system but I am unable to write over the read only file modules.

ccrisan, 2017-03-28:
@iamkeq the command is for Raspbian + motionEye, not for motionEyeOS. In motionEyeOS that happens automatically.

iamkeq, 2017-03-28:
@ccrisan first off thank you for the ridiculously fast response.  Secondly, is there a way then to add the bcm2835-v412 camera easily when motioneyeos is already installed and running?

Sorry if I am missing something here I am still a novice.

ccrisan, 2017-03-28:
@iamkeq it is added by default. It should be there. If it's not loaded, is because the camera could not be physically detected. You can chech its presence using the command `lsmod`.

trolleypoleking, 2017-03-29:
Having the exact same problem. Running motionEyeOS on a rpi3. Raspberry Pi NoIR Camera Module V2 not showing up under "add Camera..." but lsmod shows it's connected.

ccrisan, 2017-03-29:
@trolleypoleking do you have a `/dev/video0`?

trolleypoleking, 2017-03-29:
Thanks for the quick reply.
No /dev/video0 that i can see.

ccrisan, 2017-03-29:
So `lsmod` shows a bcm2835-v4l2 but you don't have a `/dev/video0`? What's the output of the commands `lsmod`, `ls /dev/video*` and `v4l2-ctl --list-devices`?

trolleypoleking, 2017-03-29:
[root@HomeEye dev]# ls /dev/video*
ls: /dev/video*: No such file or directory
[root@HomeEye dev]# v4l2-ctl --list-devices
Failed to open /dev/video0: No such file or directory
[root@HomeEye dev]# lsmod
Module                  Size  Used by    Not tainted
ipv6                  346619  0 [permanent]
sha256_generic          9327  1 
hmac                    2959  1 
drbg                   13648  1 
ctr                     3757  1 
ccm                     8240  1 
snd_soc_bcm2835_i2s     6290  0 
snd_soc_core          126350  1 snd_soc_bcm2835_i2s
snd_bcm2835            20395  0 
snd_pcm_dmaengine       3391  1 snd_soc_core
snd_pcm                75620  3 snd_soc_core,snd_bcm2835,snd_pcm_dmaengine
brcmfmac              186310  0 
snd_timer              19476  1 snd_pcm
brcmutil                5661  1 brcmfmac
snd                    52522  4 snd_soc_core,snd_bcm2835,snd_pcm,snd_timer
i2c_bcm2708             4834  0 
bcm2835_gpiomem         2972  0 
spi_bcm2835             6690  0 
arc4                    1958  2 
rt2800usb              17618  0 
rt2800lib              72217  1 rt2800usb
rt2x00usb               8879  1 rt2800usb
rt2x00lib              37289  3 rt2800usb,rt2800lib,rt2x00usb
mac80211              533369  3 rt2800lib,rt2x00usb,rt2x00lib
cfg80211              430868  3 brcmfmac,rt2x00lib,mac80211
uio_pdrv_genirq         3164  0 
uio                     8192  1 uio_pdrv_genirq
crc_ccitt               1278  1 rt2800lib
rfkill                 16134  2 cfg80211
bcm2835_wdt             3225  1 
bcm2835_v4l2           38417  0 
videobuf2_vmalloc       5187  1 bcm2835_v4l2
videobuf2_memops        1332  1 videobuf2_vmalloc
videobuf2_v4l2         16471  1 bcm2835_v4l2
videobuf2_core         20996  2 bcm2835_v4l2,videobuf2_v4l2
v4l2_common             3865  2 bcm2835_v4l2,videobuf2_v4l2
videodev              133090  4 bcm2835_v4l2,videobuf2_v4l2,videobuf2_core,v4l2_common
media                  11573  1 videodev

trolleypoleking, 2017-03-30:
Nevermind, i'm an idiot. Had the camera plugged into the display dsi.

gabe31415, 2017-04-25:
@trolleypoleking I just did the same damn thing and didn't understand until you owned up to it... thank you for sharing!

shiz1, 2017-04-28:
@ccrisan Still having issues, camera not detected in the GUI. Pi is B+, camera module is a chinese knockoff of the PiNoir. I tested it with raspistill in Rasbian and it takes photos fine, so the camera is functional. Here's the output from the commands you requested above.

[root@meye-18c9f311 ~]# lsmod
Module                  Size  Used by    Not tainted
ipv6                  320776  0 [permanent]
snd_soc_bcm2835_i2s     5979  0
regmap_mmio             2846  1 snd_soc_bcm2835_i2s
evdev                  10453  0
snd_bcm2835            19199  0
snd_soc_core          128432  1 snd_soc_bcm2835_i2s
snd_pcm_dmaengine       3996  1 snd_soc_core
snd_pcm                76635  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
snd_timer              18927  1 snd_pcm
i2c_bcm2708             4943  0
spi_bcm2835             6556  0
snd                    52443  4 snd_bcm2835,snd_soc_core,snd_pcm,snd_timer
bcm2835_gpiomem         2969  0
uio_pdrv_genirq         3146  0
uio                     7720  1 uio_pdrv_genirq
bcm2835_wdt             3218  1
bcm2835_v4l2           37283  0
videobuf2_vmalloc       5068  1 bcm2835_v4l2
videobuf2_memops        1418  1 videobuf2_vmalloc
videobuf2_v4l2         17014  1 bcm2835_v4l2
videobuf2_core         20295  2 bcm2835_v4l2,videobuf2_v4l2
v4l2_common             4279  2 bcm2835_v4l2,videobuf2_v4l2
videodev              140303  4 bcm2835_v4l2,videobuf2_v4l2,videobuf2_core,v4l2_common
media                  12111  1 videodev

france193, 2017-05-07:
Hi @ccrisan, I have A Raspberry Pi 3 and a Raspberry Pi NoIR Camera Module V2.1, after installing motioneyeOS (taken from here https://github.com/ccrisan/motioneyeos/releases/download/20170212/motioneyeos-raspberrypi3-20170212.img.gz), i read that was the last stable version, the camera was detected but doesn't work... However i tried the camera and it's working but doesn't work with motion eye...
[image]
[image]

Any suggestion? thanks a lot!

ccrisan, 2017-05-07:
Does it work with Raspbian? Did you tweak the GPU memory?

france193, 2017-05-07:
The camera with raspbian works, i read this guide and it works... https://www.raspberrypi.org/learning/getting-started-with-picamera/
i tried a lot of other guides but i never be capable of do a stream from the camera but only photo taken with python library

france193, 2017-05-07:
No, i didn't tweak GPU memory, and now i'm with motioneyeOS, and i didn't modify anything

ccrisan, 2017-05-07:
Hmm that's odd. Power supply good enough? Log files complaining about anything?

france193, 2017-05-07:
I tried this code:

from picamera import PiCamera
from time import sleep

camera = PiCamera()

camera.start_preview()
sleep(10)
camera.stop_preview()

with raspbian (with same camera) with a Raspberry Pi 2 and the camera it's working, the same with a raspberry pi zero W... With the Raspberry Pi 3 i have some problems:
[image]

So, the problem is only with Raspberry Pi3, or only with my Raspberry Pi 3 camera interface...How to be sure that the interface is broken or not? (i used the same cable for RPi2 and RPi3 and the cable it's ok).

Now i'll try motionPiOS on the Raspberry Pi Zero ...

france193, 2017-06-01:
I solved, there was a problem with the Raspberry pi 3 and i contacted the assistance to substitute the raspberry, the it works perfectly! thanks @ccrisan !

cahmedc, 2017-09-22:
@france193 how did you solve it ?!

france193, 2017-09-28:
@cahmedc I contacted the shop where i bought the raspberry and they provided me a replacing unit because the other has a problem on the camera connection cable

scotttetrick, 2017-10-03:
I am having a similar issue with the Raspberry Pi Zero W and Motion Eye with Raspbian
[image]

jasaw, 2017-10-03:
@scotttetrick Which version of motionEyeOS are you running?
Please make sure your camera flat cable is connected correctly.
SSH into your Pi Zero W and check whether /dev/video0 exists.

walkstertomo, 2017-11-11:
I have a similar problem, pi3 with csi camera, /dev/video0 gives (No such file or directory
)

ksauln, 2018-01-14:
@scotttetrick  were you able to get yours working? I'm having the same problem. I was about to take a picture using raspistill so I know the camera is connected and working

scotttetrick, 2018-01-14:
My issue was a bad/broken camera. Working now with new camera

Gladfelder, 2018-04-12:
Camera not recognised - Pi1 - running out of ideas.
I started with MotioneyeOS. No camera seen or addable. Terminal, vcgencmd get_camera, supported=1 detected=1
So the Pi sees the camera but the software does not.
After various attempts I gave up and tried installing Raspbian Stretch Lite, updated, install Motioneye.
Same as before! Maybe something missing in Lite?
I am now on  Raspbian Stretch FULL., updated, Motioneye installed. STILL NO CAMERA.
I have tried everything I can find without luck so far.

```
pip install motioneye --upgrade
<snip>
Successfully installed MarkupSafe-1.0 jinja2-2.10 pillow-5.1.0 six-1.11.0

root@Pi-CAM:/home/pi# vcgencmd get_camera
supported=1 detected=1

root@Pi-CAM:/home/pi# lsmod
Module                  Size  Used by
bluetooth             366751  0
ecdh_generic           17649  1 bluetooth
fuse                   92214  3
ctr                     4422  6
ccm                    10611  9
arc4                    2332  2
rt2800usb              18197  0
rt2800lib              95872  1 rt2800usb
rt2x00usb              12185  1 rt2800usb
rt2x00lib              47540  3 rt2800lib,rt2800usb,rt2x00usb
mac80211              650839  3 rt2800lib,rt2x00lib,rt2x00usb
cfg80211              542242  2 rt2x00lib,mac80211
rfkill                 21476  4 bluetooth,cfg80211
crc_ccitt               1654  1 rt2800lib
snd_bcm2835            23054  0
snd_pcm                89526  1 snd_bcm2835
snd_timer              22396  1 snd_pcm
snd                    59578  3 snd_timer,snd_bcm2835,snd_pcm
uio_pdrv_genirq         3718  0
fixed                   3033  0
uio                     9901  1 uio_pdrv_genirq
bcm2835_v4l2           42574  0
v4l2_common             5762  1 bcm2835_v4l2
videobuf2_vmalloc       5689  1 bcm2835_v4l2
videobuf2_memops        1802  1 videobuf2_vmalloc
videobuf2_v4l2         15667  1 bcm2835_v4l2
videobuf2_core         30680  2 bcm2835_v4l2,videobuf2_v4l2
videodev              169983  4 v4l2_common,videobuf2_core,bcm2835_v4l2,videobuf2_v4l2
media                  26672  1 videodev
i2c_dev                 6674  0
ip_tables              12395  0
x_tables               21417  1 ip_tables
ipv6                  399486  22

root@Pi-CAM:/home/pi# ls /dev/video*
/dev/video0

root@Pi-CAM:/home/pi# v4l2-ctl --list-devices
mmal service 16.1 (platform:bcm2835-v4l2):
        /dev/video0

root@Pi-CAM:/home/pi#

raspistill -o image.png  (SUCCESS! image taken, 2592x 1944 pixels)

raspivid -o video.h264    (SUCCESS!)
MP4Box -add video.h264 video.mp4 (viewable, hi-def video)

```
So, I can take stills and video from the command line but Motioneye still looks like this.

[image]
[image]

ANY help would be appreciated. Thanks.

Gladfelder, 2018-04-13:
@jasaw, sorry but NooB, how do I get these?

btong, 2018-11-27:
I had this same problem: everything was supported and recognised by the Pi, but not by motioneye. The problem was that motion was not installed (copy paste failure). /var/log/daemon.log told me this. After I installed it, everything worked.

wiskywalker, 2018-12-24:
> Nevermind, i'm an idiot. Had the camera plugged into the display dsi.

I had made the same mistake. I guess the human error is still one of the most annoying.

thewaver007, 2019-09-05:
Hi, I'm trying to get motion eye to work on my pi zero w but am having problems with seeing the camera. The camera is attached directly to the board and works when I enable 'fast network camera' in the 'expert settings'. When I turn the 'fast network camera' off, I get no picture at all. its a cheap chinese camera. Is there a setting that i need to set or am i being a complete idiot. Thanks

Stofferl, 2019-11-22:
Hello!
I know it's an old post, but i had the same problem.
"vcgencmd get_camera" shows "supported = 1 detected = 1
The same camera works on another Motioeyeos without problems.
If i enable the "Fast Network Camera" option, it's working, but not in normal mode.
The Problem was NOT MotioneyeOS or the Raspi, it was Google Chrome browser.
I tried Firefox or MS Edge and it worked.
I hope this helps someone! :-)

dborden0, 2019-11-26:
> Hello!
> I know it's an old post, but i had the same problem.
> "vcgencmd get_camera" shows "supported = 1 detected = 1
> The same camera works on another Motioeyeos without problems.
> If i enable the [...]

This absolutely helped me! Thank you!

ajones-19, 2020-01-23:
I'm having this issue as well. I am using a Pi Zero W with motioneyeOS and a camera connected to the CSI. If I use fast network camera, I get a video image. If I turn off fast network camera, it says it can't open the camera. I have tried multiple browsers on both mobile and Windows (Chrome, Firefox, Edge) and multiple camera modules.

ebolisa, 2020-09-14:
Hi,
I had to reinstall motion and I lost control over the web interface. Where should I start to look?
TIA

Port 8080 (shown in motion.conf) is streaming
Port 8081 (where the motionEye web should come up, I get what is shown in the img.

[image]

starbasessd, 2020-09-14:
It's going to be at port 80 (motionEyeOS, or port 8765 if running
motion/motionEye on another OS like Ubuntu.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

shbik, 2021-02-03:
hi i have write the motioneyeOS on my SD card but looks like it doesnt boot
i have try it on a raspberry pi 4 
can someone can help me

ebolisa, 2021-02-03:
What do you see on the monitor?

shbik, 2021-02-03:
Bootloader error screen
I’ll try to do all the processes  again

starbasessd, 2021-02-03:
You need to update theEEPROM boot loader.:
Install RaspberryPiOS on the SDCard
Boot it up.
Run the following commands:
sudo apt update [enter]
sudo apt full-upgrade [enter]
sudo rpi-eeprom-update -a
sudo reboot [enter]
After the reboot run:
sudo rpi-eeprom-update [enter]
and confirm the boot loader is dated after Sept 2020
If it is, install motionEyeOS on the SDCard and boot up normally.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

shbik, 2021-02-04:
After this commands I need to follow the motioneye install on raspbian?

starbasessd, 2021-02-04:
No, you can re-image back to motionEyeOS.
Unfortunately, you can't update the bootloader while in motionEyeOS, that
is why you need to use RaspberryPiOS, JUST for the bootloader update.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

shbik, 2021-02-04:
[image]

shbik, 2021-02-04:
[image]

starbasessd, 2021-02-04:
If you were trying to post screenshots, they didn't work

[9 more replies]