motioneye-project/motioneyeos #291: Won't login via web browser remotely

Kayak83, 2016-05-11:
I'm unable to log into the Pi camera set up on port forwarding remotely via a browser. Tested Chrome and IE. The login prompt asks for my credentials and when I select "login," it's refreshes and loops, asking to enter login credentials again. Tried both surveillance user and admin logins with no success.

I _AM _able to log in and view the camera remotely via an IP cam viewer app on my android phone. I force closed the app to log out of the camera and tested in the browser, but that didn't fix it. I thought maybe there was a max number of users able to be logged in at once.

ccrisan, 2016-05-12:
Do you use some fancy password with special characters in it? Do you use a reverse proxy? There's no   maximum limit of logged in users.

Kayak83, 2016-05-12:
PW had a special character in it. Nothing crazy. A " + ".  

And the Android app logged in just fine.

ccrisan, 2016-05-12:
You said you were using port forwarding. Does it work locally, by any chance?

Kayak83, 2016-05-12:
I was, and it was working locally on my PC. 

Locally PC (chrome) = working
Remotely Android (IP Cam Viewer) = working
Remotely PC (chrome and tested IE) = Won't log in. Login popup resets and pops back up empty when entering credentials and hitting "log in."

Now that I have you attention as well, there is a firmware update that fixes the V2 camera's mirrored image issue. Any way to implement that into motioneye? Basically just a flip image.

ccrisan, 2016-07-03:
Should be fixed.

closed 2016-07-03 by ccrisan