motioneye-project/motioneyeos #292: Flipping video options

kmckmc01, 2016-05-12:
In 'normal' camera mode my video stream is mirrored/reversed. With Fast Network Camera option set there is an option to flip the video horizontally which is not there in normal mode.

I'm using a Rpi 3 and camera v2. Would be grateful for any advice or comment.

mchwalt, 2016-05-23:
Same to me:
Raspi 1B and camera V2.1 => sides are inverted

waxlabo, 2016-05-25:
I am also experiencing this issue.  It would be great to be able to apply horizontal flip to the camera in normal mode. :)

mehtaman, 2016-05-26:
I have just switched over to a Pi3 + V2 NoIR and noticed the videos are mirror imaged.

I tried to follow these steps but they did not work for me:
@cjmair

> I would try this.
> SSH to the Pi, as root, then use these commands
> 
> CD /etc
> mount -o remount,rw /
> nano rc.local
> 
> Use the editor to add the following two lines:
> 
> #!/bin/sh
> v4l2-ctl --set-ctrl horizontal_flip=1
> 
> use Ctrl&O to save the file then Ctrl&X
> issue a reboot command
> 
> Good Luck

Post: [https://github.com/ccrisan/motioneyeos/issues/278](https://github.com/ccrisan/motioneyeos/issues/278)

gavros, 2016-05-26:
If you are using the motioneyeos image, you can fix this by doing the following:

ssh into your device
create this file using the nano command:
nano /data/etc/userinit.sh
add the below line into the userinit.sh file
v4l2-ctl --set-ctrl horizontal_flip=1
save and reboot your device.

waxlabo, 2016-05-26:
@gavros awesome, thanks so much pal!  That did the trick.

mchwalt, 2016-05-29:
@gavros: that's it !  thans a lot :-)

ccrisan, 2016-06-05:
The new OS image with RPi Zero and camera v2.1 support will be available later today.

closed 2016-06-05 by ccrisan