motioneye-project/motioneyeos #294: Upload to google drive stops once admin password is set

juntek, 2016-05-15:
Uploading captured movies to google drive works only when admin password has not been set.
Once I set the password motioneye stops uploading movies.

ccrisan, 2016-05-16:
Please attach your `moitioneye.log` file.

juntek, 2016-05-17:
Sorry for delay, log attached. Pretty much normal, reboot at 21:09:33 relates to admin password setting.

[motioneye.log.zip](https://github.com/ccrisan/motioneyeos/files/268978/motioneye.log.zip)

ccrisan, 2016-05-18:
There's nothing suspicious in your log file. If you change your admin password back to what it was before, do your media files get uploaded to Google Drive?

juntek, 2016-05-19:
Yes, upload to Google Drive is resumed once I bring empty admin password back.

ccrisan, 2016-05-19:
Can you please enable Debugging in Expert Settings, change your password again so that it won't work, leave it running as much as needed for some media files to be created and send me the motioneye.log file again?

juntek, 2016-05-19:
I think I've found the culprit. Upload to Google Drive doesn't work if admin password contains a $ sign. I've tested it with the two passwords: admin1234 and admin$1234

ccrisan, 2016-05-19:
Good to know. I'll look into it and get back to you asap.

ccrisan, 2016-05-20:
Fixed with https://github.com/ccrisan/motioneye/commit/2890653304c1e5dedd2d1c4b64cb92903d0a7fbc.

closed 2016-05-20 by ccrisan

rushless, 2016-05-21:
Hi, 

I had this issue also, this thread helped me thank you. In my case I had an @ sign that was causing the issue not a $. Thought it might help you troubleshoot.

ccrisan, 2016-05-21:
The issue was valid for all special characters.