motioneye-project/motioneyeos #296: Images not attached to email notification

Gryglas, 2016-05-16:
Hi.
I have activated the motion notifictaion email to on.
Set up the smtp server and it sends me notification emails now.
Also enabled still images and set the Capture Mode to Motion Triggered. And I have made sure to configure Attached Pictures Time Span to a reasonable amount of 10 seconds.

At the same time the LED on/off isn't switching the Raspi Camera LED on or off. So I'm thinking there is something wrong with motionEyeOS all together?

best regards and thanks for great software!
/Håkan

ccrisan, 2016-05-16:
Please attach your log files (all of them). If you can log in remotely to your PI, please also attach here the content of `/boot/config.txt`.

Gryglas, 2016-05-16:
Hi!
Thanks for quick reply. Attached are the log files.
[Arkiv.zip](https://github.com/ccrisan/motioneyeos/files/266686/Arkiv.zip)

Gryglas, 2016-05-16:
sorry, i closed by misstake.

Gryglas, 2016-05-18:
Hi ccrisan. Any progress? How can I with read AND write permission change the config.txt in terminal? Can't seem to use SUDO.

ccrisan, 2016-05-18:
You don't need sudo. You already are root. You'll always be on motionEyeOS. However you just need read permissions. Just use `cat /boot/config.txt` and copy the contents. What camera do you use by the way? Is it the old standard one?

Gryglas, 2016-05-19:
Hi.
Well, it’s one of the first raspi cams but I thought they all where the same?
Here are the contents of the config.txt
[root@meye-4d7781c3 ~]# cat /boot/config.txt
start_file=start.elf
fixup_file=fixup.elf
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
disable_camera_led=1
device_tree=bcm2710-rpi-3-b.dtb

What do you want me to do with this now?

best regards,
Håkan

ccrisan, 2016-05-19:
You have the `disable_camera_led=1` option. That means motionEyeOS did its best to disable the camera led. If the camera led is still on, the camera model is probably not 100% compatible with motionEyeOS. Not al raspi cams are identical. You have at least 4 official variants (v1 normal, v1 noir, v2 normal, v2 noir) and a whole lot of non-official ones that still claim some compatibility.

Gryglas, 2016-05-19:
Hi Calin.

Ok, I see.
But I have the official v1 normal. Should this not be compatible with motionEyeOS?

I still can't seem to edit the config.txt anyway. When i run:
ssh admin@10.0.0.37
vi /boot/config.txt
i (press i key for edit)
now the terminal says "read only" so I can't make any changes to it. :(

best regards,
Håkan

ccrisan, 2016-05-19:
It definitely is. If the camera led is still on in these circumstances, then I have no idea what the problem could be.

Gryglas, 2016-05-19:
Ok. Then I'll have to look around.
But how do I Edit the files in the os?
Do you use a ftp client for download, edit them and upload them again or do you use terminal? I still can't seem to change files. :(

ccrisan, 2016-05-19:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file

Gryglas, 2016-05-19:
Hi Calin.

I found this: https://www.raspberrypi.org/forums/viewtopic.php?p=922799#p922799 <https://www.raspberrypi.org/forums/viewtopic.php?p=922799#p922799>
So the LED on the pi3 has a problem with the firmware.
Can I somehow do a ”sudo apt-get update” on the motioneyeos in terminal?

best regards,
Håkan

https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file <https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file>

ccrisan, 2016-05-20:
You can't manually update parts of motionEyeOS. I will make sure the latest versions of everything will be bundled with the next release.

Gryglas, 2016-10-11:
Hi.
So after a couple of upgrade releases of your software the LED-turn off-turn-on-function still doesn’t work on the raspberry pi 3 I’m afraid. Is this just something you have missed out in your releases?

kind regards,
Håkan

ccrisan, 2016-10-11:
What camera module do you have?

Gryglas, 2016-10-11:
The original raspberry camera module.

ccrisan, 2016-10-11:
Which one? What version?

Gryglas, 2016-10-11:
The first version, 5 megapixel.

ccrisan, 2016-10-11:
Does the LED work with Raspbian?

Gryglas, 2016-10-11:
Yep.

ccrisan, 2016-10-26:
I'm sorry but I wasn't able to reproduce the problem.

Gryglas, 2016-10-27:
Ok. I'll try to format and reinstall then and see if that helps. Thank you for testning.

closed 2017-06-01 by bortek