motioneye-project/motioneyeos #297: Can't seem to correct the Mirror Image

ryanpwest, 2016-05-16:
I have the Raspberry Pi 3 and the Camera Module V2 and the image is appearing mirrored.
I cannot find an option in the settings to enable/disable mirrored image.

I came across the following thread about putting a command in rc.local but rc.local never appears to run on its own.
https://github.com/ccrisan/motioneyeos/issues/278

b0rnwr0ng, 2016-05-20:
I've been having this problem, too and came across this--  following the last two posts on that thread worked for me (adding to the S99showinfo file)

gavros, 2016-05-26:
If you are using the motioneyeos image, you can fix this by doing the following:

ssh into your device
create this file using the nano command:
nano /data/etc/userinit.sh
add the below line into the userinit.sh file
v4l2-ctl --set-ctrl horizontal_flip=1
save and reboot your device.

closed 2017-06-01 by bortek