motioneye-project/motioneyeos #298: stop pi from rebooting to correct wifi setting

ts4s, 2016-05-19:
How does one stop the pi from auto rebooting when it cant connect to network? i've tried connecting to my router via ethernet and its not even looking at the port for an ip address.I've tried direct connect to my mac with an assigned IP address, no luck. Tried to be quick on the login and stop the process but the kill ### doesn't work. really don't what to have to re format and config it all again.

ccrisan, 2016-05-19:
Check out the Expert Settings configuration section and disable the network watch mechanism.

ts4s, 2016-05-19:
can't access that screen if the pi is not on the network and is constantly rebooting itself. I have an ethernet cable plugged in but when it boots, it's only looking for the wlan. So I'm trying to login locally on the pi to stop the 20 second reboot inorder to correct the network setting. does the kill command not work in this os?

ccrisan, 2016-05-21:
I made sure the next version won't reboot your PI when WiFi is configured but not available, while ethernet is connected.

closed 2016-05-21 by ccrisan