motioneye-project/motioneyeos #299: Images corrupting after a day or two.

captaintapir, 2016-05-19:
Hi,

I've been using MotionPie/motioneyeos for over a year, and it's been great.  I've recently hit an issue with images being saved - after a day or so after rebooting, they become corrupt (such as the example below).  The live view looks fine, and soon as I reboot it clears the issue up for another day or so until it happens again.

[image]

Nothing's changed with the configuration to cause the problem.  The only relevant possible errors I can find in the logs are these that must have occurred around the possible times that the issue started most recently (but without an image, it's hard to be sure!):

motioneye.log
2016-05-17 21:01:00: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed

messages.log
May 18 17:55:50 meye-******** user.err motion-mrdave: [1] [ERR] [VID] v4l2_next: VIDIOC_QBUF: No such device
May 18 17:55:50 meye-******** user.err motion-mrdave: [1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
May 18 17:56:00 meye-******** user.warn motion-mrdave: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
May 18 17:56:00 meye-******** user.warn motion-mrdave: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00Jan  1 10:00:09 meye-******** auth.err passwd: /etc/passwd: Read-only file system
Jan  1 10:00:09 meye-******** auth.err passwd: can't update password file /etc/passwd
Jan  1 10:00:09 meye-******** auth.err passwd: /etc/passwd: Read-only file system
Jan  1 10:00:09 meye-******** auth.err passwd: can't update password file /etc/passwd

(the time appears to have reset to 1.1.1970, although it doesn't appear to have affected the timestamp on corrupt images)

I'm thinking to have the camera reboot automatically daily, but that's a workaround and not a solution, so any ideas on how to fix the problem would be great, please.

ccrisan, 2016-05-19:
It looks like a hardware problem, most likely coming from the camera. What camera do you use?

captaintapir, 2016-05-19:
Thanks for the quick response.  It's a Microsoft webcam that's been decased and mounted inside another housing with the RPi (LifeCam VX-700, from memory).  I did wonder about that, but seeing that restarting it does fix the problem every time, I had started leaning away from this being the cause.

I've been loathe to pull it apart to check as it's mounted in a high position but it may be time for an upgrade, in this case.

ccrisan, 2016-07-03:
This most likely looks like a bug in the camera driver. There's nothing I can do about it.

closed 2016-07-03 by ccrisan