motioneye-project/motioneyeos #303: Central server (ubuntu) to do the motion

freyagobi, 2016-05-22:
Hi,

I have a rasPi3 using motioneyeos Version 20160410 (great software + web interface by the way).

Under "expert settings" I have enabled "Fast Network Camera".

I have a server running ubuntu with motionEye running (following the steps on https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu).

As I understand it (correct me if I'm wrong), you can use the server to do the motion detection + recording (so the pi only streams) thus letting the server do all the hard work.

In my "Video Streaming" settings on the Pi, I have:

Streaming URL: http://192.168.1.241:8081/
Embed URL: http://192.168.1.241/picture/1/frame/
Streaming Port	 : 8081

If I go to my server via http://192.168.1.230:8765/, I can add the Pi via:
Camera Type = Simple MJPEG Camera
URL = http://192.168.1.241:8081/
Camera = MJPEG Network Camera

After a few seconds the camera shows up in the server browser (http://192.168.1.230:8765/) but no motion detections or recording options are available.

If turn off "fast network camera" and I add the camera as a "Remote motionEye Camera", using:
Camera Type = Remote motionEye Camera
URL : http://192.168.1.241:80
I can set motion etc but (again correct me if i'm wrong), the Pi is doing all the motion detection work not the server (plus files are stored on the pi).

Looking over https://github.com/ccrisan/motioneyeos/issues/40, from what you and users have mentioned, It sounds like you can set it up to the server does all the ard work.

My question is, how do you set the server up to do all the motion and the Pi's just stream the video?

ccrisan, 2016-05-23:
On your server, add the camera as a Network Camera instead of Simple MJPEG Camera. The rest of the settings should be the same.

freyagobi, 2016-05-23:
Hi,

I'm not sure how to add the camera as a network camera in motioneye. The only dropdowns I have from the add camera dropdown is  " Remote motionEye Camera "and " Simple MJPEG Camera ".

ccrisan, 2016-05-23:
In that case you don't have the motion daemon installed on your server.

freyagobi, 2016-05-23:
Fixed, rebuilt server and re-did the steps and motion installed fine. Worth noting that motion detection doesnt show up unless advanced settings are enabled.

Thanks for you help and great software.

closed 2016-05-23 by freyagobi