motioneye-project/motioneyeos #304: No computer name in Samba

Petreleslie, 2016-05-22:
Finally got the Motioneyeos camera working with my network storage. thanks for all your good work and for continuing support. (Made a donation today)

The network storage works with my second Raspberry Pi as a NAS but only with its IP address. I would like to be able to use the NAS computer's name. Windows computers on the network see it but the Motioneyeos Raspberry doesn't work with the name only the IP address.

Does samba need some addition config to make this work? 

[image]

And is there a way to make the a larger camera preview on the right pane?

Thanks for sending this newbie in the right direction.

ccrisan, 2016-05-23:
It should work with the name as well. How do you know it doesn't work with the NetBIOS name?

P.S. Check the preferences/layout in order to adjust the frame size.

Petreleslie, 2016-05-23:
Preferences! Who knew? Thanks. 
BTW the help files must not be installed correctly for me. Clicking on any question mark shows nothing.....
The screenshot shows what happens with the NetBios name. This is the name in the Windows machine and from running nbtstat at a Windows command line. I've tried adding a forward backslash and also leaving it off, giving the same result. 

[image]

closed 2016-05-23 by Petreleslie