motioneye-project/motioneyeos #305: Error message with Email Notifications

kennyken444, 2016-05-22:
Hi, I've been playing around with MotionEye and really like it well done.
However I was just having an issue that i was hoping you could help me with.

I can see the camera working fine when i browse to it in a browser, in the settings i have turned on motion detection and the notifications. 
I am running a raspberryPi 2 with two standard raspberry Pi cameras a V1 and V2 (not at the same time, just testing each one) and i seem to keep getting this error message:
v4l2_select_input: Device doesn't support VIDIOC_G_STD
and i never get any emails sent though. from reading through this area, it looks as though my setup is supported so i guess i must have something else wrong. Any advice?

[Motion log.txt](https://github.com/ccrisan/motioneyeos/files/276582/Motion.log.txt)
[Motioneye Log.txt](https://github.com/ccrisan/motioneyeos/files/276583/Motioneye.Log.txt)
[dmesg log.txt](https://github.com/ccrisan/motioneyeos/files/276584/dmesg.log.txt)

Thanks in advance
Ken

ccrisan, 2016-05-23:
You can safely ignore the VIDIOC_G_STD error. Does your setup detect motion (i.e. does the camera frame ever turn red)?

kennyken444, 2016-05-23:
Thanks for getting back to me, Yes i have seen the frame definitely turn red when i walk passed the camera.

Ken

kennyken444, 2016-05-23:
[Screen Shot 2016-05-23 at 22.22.48.pdf](https://github.com/ccrisan/motioneyeos/files/278292/Screen.Shot.2016-05-23.at.22.22.48.pdf)

ccrisan, 2016-05-24:
And what do your email/smtp settings look like? Can you send a screenshot with them as well?

kennyken444, 2016-05-24:
[image]
Here are the SMTP Settings. to be honest i was focussing on the original error and didn't look at my smtp settings but i think they are correct. i am assuming i can use my hotmail account...

Thanks
Ken

ccrisan, 2016-05-25:
Things look alright to me. The funny thing is that I don't recall seeing anything about email in your log files. Could it be possible that you gave me the log files corresponding to a period where no email would have been sent? In any case, please try reproducing the problem again (i.e. make sure the camera frame turns red and that email notifications are enabled) and please collect and send me the logs again.

P.S. A test email button is long overdue :)

kennyken444, 2016-05-26:
[Motioneye-Thur.txt](https://github.com/ccrisan/motioneyeos/files/285131/Motioneye-Thur.txt)
[dmesg-thur.txt](https://github.com/ccrisan/motioneyeos/files/285136/dmesg-thur.txt)
[Motion-Thur.txt](https://github.com/ccrisan/motioneyeos/files/285129/Motion-Thur.txt)
Thanks Chap, a Test email button would be great :-)
Attached are 3 log files. I  made sure the the image border went red for at least 30 seconds i think and then waited a couple of mins and downloaded the log files.
Fingers crossed this highlights something

Thanks again for your help
Ken

ccrisan, 2016-05-26:
I'm sorry but there's no trace of email sending in your log files. The only thing I can do to help at this point is to ask for your config files. Please make sure you have removed/replaced all sensitive data (i.e. passwords) before sending them. I'll try to reproduce then the issue myself.

kennyken444, 2016-05-26:
How weird - I'm sure I've ticked everything i was supposed to, to get it to try and send an email when motion is detected.

Which files do you want me to send that contain the config?

Thanks
Ken

ccrisan, 2016-05-27:
I need your `/data/etc/motion.conf` and `/data/etc/thread-*.conf` files. Again, don't forget to remove your passwords.

kennyken444, 2016-05-27:
[thread-1.txt](https://github.com/ccrisan/motioneyeos/files/286977/thread-1.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/286978/motion.txt)

Thanks, i have changed the password :-)

kennyken444, 2016-06-02:
Any chance you've had a look at my config files? 
Thanks
Ken

ccrisan, 2016-06-05:
@kennyken444 do you have any special characters in your actual admin password? If yes, could you try setting a simple password and see if the problem persists?

kennyken444, 2016-06-06:
Do you mean in your software or for my email account? Eitherway i think the answer is no because i havent set an admin password, its still the default one in MotionEye and for my email there is a capital and a number but no special characters. 

Thanks
Ken

ccrisan, 2016-06-06:
I mean motionEye's admin password. Anyways could you please try the latest prerelease version and see if the problem persists? I have done some fixes in this direction/

kennyken444, 2016-06-06:
Not sure if it was the new version or changing the sending SMTP account to a GMail one but that has got it working, maybe hotmail just doesnt like it. 
Thanks again for all your help :-) 
I hope the small donation i made helps a little

ccrisan, 2016-06-06:
Thanks! I'll close the issue then.

closed 2016-06-06 by ccrisan