motioneye-project/motioneyeos #307: Cannot detect camera on Pi Zero v1.2

kieranc, 2016-05-24:
I've installed MotionEyeOS 20160410 on a new Pi Zero with a standard 5MP Pi camera, but I can't get the camera to work. If I boot Raspbian on the same hardware (except the USB wireless, trying to test meye without wifi) it works, so I'm fairly sure it's not the cabling/hardware.

[root@meye-b35dc5c1 ~]# modprobe bcm2835-v4l2
modprobe: can't load module bcm2835-v4l2 (kernel/drivers/media/platform/bcm2835/bcm2835-v4l2.ko): Operation not permitted

[root@meye-b35dc5c1 ~]# vcgencmd get_camera
supported=0 detected=0

[root@meye-b35dc5c1 ~]# vcgencmd get_mem gpu
gpu=128M

[root@meye-b35dc5c1 ~]# cat /boot/config.txt 
start_file=start.elf
fixup_file=fixup.elf
#arm_freq=700 <-- worth noting this is incorrect for the Zero by default
#core_freq=250
#sdram_freq=400
#over_voltage=0
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
disable_camera_led=1
#start_x=1

[root@meye-b35dc5c1 ~]# dmesg | grep v4l2
[    2.638543] : bcm2835-v4l2: error -1 while loading driver

gavros, 2016-05-25:
Same issue here. It also not working with the 8MP v2.1 camera module as well (MotionEyeOS 20160410). Just wont detect  the camera

madromark, 2016-05-27:
Same here, with a new Raspberry Pi Zero, 5mp camera module and the newest Motioneyeos.

moritzmhmk, 2016-05-29:
same here (zero v1.3 and camera v2.1) - can I be of any help fixing this?

ccrisan, 2016-05-29:
I'm working on it.

AdamMalkin, 2016-06-01:
Bump, I am having the same issue on a zero 1.3 with the official pi camera.

havensal, 2016-06-03:
Posting just to watch for an update.  Thanks.

kieranc, 2016-06-04:
Has anyone else tried building an image for the zero recently? I've tried twice but it won't boot, stays on the rainbow splash screen, wondering if its my setup or something else.

AdamMalkin, 2016-06-04:
I was able to do it last week, it booted up ok.

I had a problem with another os where the zero wouldn't boot when a specific os and sd card were together.
Try a different brand/size/speed of sd card

ccrisan, 2016-06-05:
A new OS image with Zero and camera v2.1 support will be available later today.

closed 2016-06-05 by ccrisan

AdamMalkin, 2016-06-05:
Great news, looking forward to trying it out :)

havensal, 2016-06-05:
@kieranc IIRC I had to use the first RPI image, not the 2 or 3.

havensal, 2016-06-06:
@ccrisan  Thank you!  After fighting my complete lack of knowledge of compiling an image, I finally got my Zero running your OS.  Most the information I needed was found here and I found the rest with a little help of Google.  Everything is configured and sending out email and uploading to Google drive.

ccrisan, 2016-06-06:
@havensal I'm glad to hear that.

AdamMalkin, 2016-06-06:
the new build works great for me, thanks

is there a build guide that handles the zero? i found it a challenge not having a wired network, i ended up doing the first boot on a b+ then moving the sd to the zero once wifi was working

madromark, 2016-06-07:
@AdamMalkin on the installation page (https://github.com/ccrisan/motioneyeos/wiki/Installation) you can find the writeimage.sh and with it you can set your wifi before booting up the Pi. I just did this on my Zero and it worked great. Thanks again @ccrisan

ccrisan, 2016-06-07:
And there's [Troubleshooting Raspberry PI Models A (Plus) And Zero](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-%28Plus%29-And-Zero).

raspberrycoulis, 2016-06-11:
I'm having trouble getting my setup to detect my camera too. I am using the latest PiZero image (20160410 found [here](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices)) but MEOS cannot detect the camera.

I'm using the PiZero with the camera module (obviously) and the older Camera Module (Rev 1.3) - i.e. not the new, rounded edge, higher resolution one.

Any suggestions?

kieranc, 2016-06-11:
20160410 is not the latest version, download 20160605 and it will work. Also, this is a closed issue, not terribly visible....

raspberrycoulis, 2016-06-11:
And where do you download this image from? The "supported devices" page only has version 20160410 images...

kieranc, 2016-06-11:
The releases page?
https://github.com/ccrisan/motioneyeos/releases

raspberrycoulis, 2016-06-11:
Ah, thanks. I've always thought the support devices page (https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices) was the downloads section!

ccrisan, 2016-06-11:
That is the correct page for supported devices. However, the Zero is only supported by a prerelease, so it didn't make it to the official supported devices list yet :)